OctoPrint/OctoPrint #1269: [Request] Include Webcam HLS source

Kriechi, 2016-03-10:
Right now the webcam can only be included as `<img>` tag - which means Motion-JPEG.
Using mjpeg-streamer, this works quite well (most of the time) - but creates an insane amount of network traffic due to uncompressed, full-frame image transfers.

Since I also want to check my printer when I'm on the road or in a waiting hall from my phone, I setup ffmpeg to provide an HTTP Live Stream. This means I need to have a `<video>` tag to include my webcam stream.

Ideally OctoPrint would support both: `img` and `video` tags,
or just make a generic option to include some arbitrary html code - that way the user can decide what do present (iframe? flash-based streamers?)
Only caveat with HLS is, that is only works in the Apple universe - so we might need to use [videojs](https://github.com/videojs/videojs-contrib-hls) to support Firefox/Chrome/Edge as well.

HorstBaerbel, 2017-07-06:
Setting up the whole HLS streaming stuff is very complicated. I think a cross-platform option would be WebRTC with x264, as "the Pis" most people use can do x264 hardware encoding OR a good JS video player solution supporting RTP/RSTP, as it is easy to use. See my experiments with an RPi zero [here](http://lektiondestages.blogspot.de/2016/12/stream-hardware-encoded-h264-video-from.html). The takeaway is that x264 can reduce the data rate needed by about 40-50%, but increases CPU usage to ~25% (versus the 2% mjpegStreamer needs).

HorstBaerbel, 2018-10-01:
FFMPEG should be able to do hardware encoding on the RPI via OpenMAX if compiled correctly (--enable-omx --enable-omx-rpi). Non-accelerated the CPU load will probably be prohibitively high...

johnboiles, 2018-10-01:
Exactly. I’ve used the OpenMAX encoder and it works pretty well.

Or in my case, my Logitech c920 can output 1080p in h264 directly. No need for any encoding on the pi, just muxing in ffmpeg.

That’s why I think if we were to do this, probably just a spot to enter an ffmpeg command would be enough. Maybe we provide a few common examples. It would be overly complex to try to provide a UI for ffmpeg’s settings

johnboiles, 2018-10-01:
Do plugins have the ability to swap out the camera viewer in the UI? Or would this require a change to the core code?

johnboiles, 2018-10-01:
We probably would want to make sure we write chunks to a RAM disk also. I could see this wearing out an as card fast otherwise

foosel, 2019-02-07:
#2001 is now the central feature request for extending the current webcam embedding by other options than just an `img` tag.

closed 2019-02-07 by foosel