OctoPrint/OctoPrint #1270: Plugin will not load cura settings file from version 17.10

uchobby, 2016-03-10:
I read that this plugin does not work with the ini files from the later versions of the stand alone version of Cura.  That seems like a significant bug!

I am trying to get OctoPrint working with a printers and can not do so because I can not load a profile. I'm totally blocked now.

In case someone has a shortcut for me.  I am trying to setup a Fabrikator Mini and can not find a compatible ini file. I have Cura 17.10 with settings but the export to ini does not work with your plugin on the current OctoPrint.

When will this be fixed? What can I do to help?

135erHeli, 2016-03-10:
Use an Older Version of Cura and Save the Settings from there

foosel, 2016-03-10:
The plugin works with Cura 15.04 as it states. Use that version and everything works. Alternatively slice on your desktop on upload the result.

closed 2016-03-10 by foosel