OctoPrint/OctoPrint #1277: [Request] On the fly gcode changes

slevin22, 2016-03-13:
Currently, if a change is necessary mid print, you can either start over or figure out where you were in the gcode file and doctor a new gcode file to pick up where the old one left off. 

What I would like is a way to modify the gcode mid print. The simplest way I can think of is an option to swap the file and have octoprint get as close as possible (even if it's just by starting on the same layer or the next layer) to where you canceled.

Ideally, although I would imagine this is much more complicated, I would like to be able to define an area in the gcode viewer and either tell it not to print there, or have it set a new speed, extrusion multiplier, etc for that area. This would allow easy cancellation of a single botched object in a print to avoid wasting filament and getting the stray filament on other objects.

ntoff, 2016-03-14:
>I would like to be able to define an area in the gcode viewer and either tell it not to print there, or have it set a new speed, extrusion multiplier, etc for that area.

That is more the job of the slicer. All octoprint's job should be is to send code to the printer. It's up to the slicer to set speeds for sections since different sized objects require longer minimum layer times in order to cool. If you were to tell Octoprint to speed up a certain area there's a good chance you'll end up with a molten mess, or spaghetti.

slevin22, 2016-03-14:
Thanks for your input!

I do understand that the initial settings should be defined by the slicer,
but should something go wrong during the print I would really like to be
able to cancel just one object or area that would otherwise cause problems
for the rest of the print. Or say one object isn't bridging properly. I
think it'd be nice to be able to tweak the speed just for that object.

gddeen, 2016-03-14:
That’s only octoprint’s job??? Whoops. Time to switch to something else! (jk)

I do see the word [request] above… And Octoprint “Does” send jobs to a slicer.

So it would be nice!!!   Excellent request.

foosel, 2016-03-14:
I'm leaving this open for now, but I seriously doubt the feasibility of this. The problems start with identifying the parts of the GCODE that needs to be changed (if it's not using absolute movements you need a whole simulation to get stuff hopefully identified), continues with parameters like extrusion width (without knowing the parameters for which the file was sliced, because the only reliable information I can extract from a GCODE file is the movement commands) and infill settings (I can't even reliably distinguish between infill, perimeters and support) etc. You can't do layer tracking since there's no reliably way to identify individual layers (z height doesn't work if z-hop is in play, or if you are in continuous mode), byte tracking won't help you if you have separate files, etc etc etc. This is one of those problems that look easy for a human to solve, but are damn near impossible to get right for a machine (unless that machine is the slicer which still has more data available than the host, which - again - only can depend on what fell out of the slicer which is basically "move to here and squirt out some plastic"). 

In any case, it will be nothing I'd want to see integrated in the core code because it would probably be rather impossible to maintain and get to work reliably. Hence labeled "plugin-only". And - no label for that but - not by me ;)

gddeen, 2016-03-14:
True. I think most people are just looking for some way of getting something to work how they imagine it could work
and getting the big people involved in doing it.

Yes. Octoprint would probably want to invoke a plugin to do this.  That plugin would
work within an environment designed to move from their design tool the slicer to octoprint,
and back to the slicer.  Slicer does this stuff already right? Not with “a blank code and no input
from a user.”

So my assumption is user wants to handle this error. They put it as a request to octoprint. 
Octoprint closes it because they’ve heard it a hundred times before.

So, maybe [barinstorming] should have been added and a bunch of people that could
get a handle on all of the work flows would be better (If that isn’t already completely done, which
I think I read before for slic3r integration). I dunno.

Great product Octoprint… Thanks.

slevin22, 2016-03-15:
Ah, well that is a bummer! Thank you very much for explaining the difficulty to me though; I'd been wondering why similar features don't exist in any gcode sender I've looked at.

Thanks for all your hard work!

ntoff, 2016-03-16:
@foosel 

>I'm leaving this open for now, but I seriously doubt the feasibility of this. The problems start with identifying the parts of the GCODE that needs to be changed 

@slevin22 
printrun actually has the ability to edit gcode in a plater environment so you can add/remove/reposition stuff. I don't know if it's feasable to add that kind of thing to octoprint though and I'm not saying it's something I'd personally want or use but just letting you know it *is* possible to edit it in a graphical environment using the same kind of plater system you'd use when initially positioning the STL files.

I don't see why simply removing a section on the fly would affect the actual movements since it's just not going to that place anymore but it would probably affect things like optimized starting points, retractions, support material and not to mention if there were any resets of the extruder in that section then it may just screw up everything. I've noticed some slicers will reset the extruder value quite frequently while others will never reset it, or only reset it here and there (the differences in absolute vs relative extrudes aside).

So theoretically you could potentially install printrun on the pi and use that to edit the gcode file that octoprint has loaded but foosel will have to confirm or deny whether octoprint loads and locks the current print file or just live streams it so whether any changes made would filter through or not.

slevin22, 2016-03-20:
@ntoff 
Thanks for the idea! I think the issue with using printrun and just swapping the file is that Octoprint (if it does not cache the file) would just pick up at wherever it was on the last file. So say it was on line 10 when the file swapped. The new file which now contains less commands may be way ahead of the old file by line 10. This goes back to the issue of figuring out where to pick back up that foosel was talking about.

> You can't do layer tracking since there's no reliably way to identify individual layers (z height doesn't work if z-hop is in play, or if you are in continuous mode), byte tracking won't help you if [...]

paukstelis, 2018-05-01:
Sorry if this is a bit necro, but I was thinking about something like this as well. If there were clear comment delimiters in the gcode (I'm think the process change comments from Simplify3D) would it not be relatively straight forward to to tell octoprint to read-ahead to the next delimiter and begin queuing at that line number?

paukstelis, 2018-05-02:
So I did a wee bit of testing, with the idea that if you have some kind of delimiter you could tag specific blocks of gcode to be ignored. This way if a single object begins to fail in a print, you could add it to a list of ignored blocks during streaming from the server. Another function might be added to skip to the next block (for example, with sequential printing). For my test I just made a few modifications in the getNext method  of comm.py:

```
			try:
				offsets = self._offsets_callback() if self._offsets_callback is not None else None
				current_tool = self._current_tool_callback() if self._current_tool_callback is not None else None

				processed = None
				skip = False
				while processed is None:
					if self._handle is None:
						# file got closed just now
						self._pos = self._size
						self._done = True
						self._report_stats()
						return None, None, None

					# we need to manually keep track of our pos here since
					# codecs' readline will make our handle's tell not
					# return the actual number of bytes read, but also the
					# already buffered bytes (for detecting the newlines)
					line = self._handle.readline()
					self._pos += len(line.encode("utf-8"))

					if not line:
						self.close()
						
					#check to see if line is '; process (.*)'
					m = self.PROCESS_RE.match(line)
					if m:
						process = m.groups()[0]
						#if it is in a list of blocked processes
						if process in self.blocked_process:
							skip = True
						#if it is not, set skip false
						else:
							skip = False
					if not skip:	
						processed = self._process(line, offsets, current_tool)
				self._read_lines += 1
				return processed, self._pos, self._read_lines
```
My skills are limited, but I'm sure there is a more benign way of farming this out to some call back that could be tapped by a plugin where a user could define their own regex match. Ideally, you could also get the block names during the gcode analysis phase, so you could have a nice popup window that would let you delete a block during a streaming print.

foosel, 2018-05-03:
There's an ongoing discussion about this on [the forum](https://discourse.octoprint.org/t/wishlist-cancel-one-of-many/1423/). You might want to check it out.