OctoPrint/OctoPrint #1279: 0.01mm z step on control page

lmclaren7000, 2016-03-14:
With some printers able to do layers at 0.1mm layers,  setting the zero with 0.1mm steps is too big a step.

0.01 would allow fine manual adjustment.

thanks

ntoff, 2016-03-14:
You can add your own custom buttons that allow you to move up and down by any amount you want.

lmclaren7000, 2016-03-14:
Yes, but it won't be in the same group as the rest of the controls unless I
misunderstand

RichardPotthoff, 2016-08-29:
I was going to ask for the same feature, but then I saw that somebody else had already put a request in.

I have modified the `control.js` file myself to add the 0.01mm button to the controls. Here is the diff-file:
```
--- a/src/octoprint/static/js/app/viewmodels/control.js
+++ b/src/octoprint/static/js/app/viewmodels/control.js
@@ -26,7 +26,7 @@ $(function() {
         self.extrusionAmount = ko.observable(undefined);
         self.controls = ko.observableArray([]);
 
-        self.distances = ko.observableArray([0.1, 1, 10, 100]);
+        self.distances = ko.observableArray([0.01, 0.1, 1, 10, 100]);
         self.distance = ko.observable(10);
 
         self.tools = ko.observableArray([]);
```
The problen is, that my changes are overwritten each time OctoPrint is updated.
So have to re-apply the patch, and recompile each time that happens:
```
cd OctoPrint
git stash pop
../oprint/bin/python setup.py install
git stash
```
It is important to "stash" the changes at the end to keep the git working directory "clean", otherwise the next automatic update may fail.
I also have two additional buttons defined: One that sets the home coordinates with a sheet of paper between the print surface and the nozzle, and another one that moves to this position, i.e. one paper thickness above true zero (may be not quite "true": I added 0.02mm to the paper thickness, so my zero is actually set 20&micro;m below the print surface). Here are the lines from the `~/.octoprint/config.yaml` file:
```
appearance: {}
controls:
- children:
  - command: G92X100Y100Z0.12
    name: Set Home
    type: command
  - command: G0X100Y100Z0.12
    name: Go Home
    type: command
  layout: horizontal
  name: Home
  type: section
feature: {}
```
My home position is in the center of the bed (x=100, y=100), and I am not using the end-stops for homing at the beginning of a print, because that would undo the manual adjustment.
I am also setting the "z-offset" setting in the slicer to 0.00mm.

foosel, 2017-01-27:
This actually sounds like it's a duplicate/different form of #162

closed 2017-01-27 by foosel