OctoPrint/OctoPrint #1280: error connecting to a replicator dual

amanda6811, 2016-03-14:
i have a Duinotech l4090 with is a copy of the replicator dual and get an error when i try to connect to my printer.

What do i need to do to get it working 

Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: <GPX.gpxprinter.GpxPrinter instance at 0x16bb558>, starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110*35
Recv:  Sailfish v7.6
Send: N0 M110*35
Recv: echo: gcode to x3g translation by GPX
Recv: SD card ok
Changing monitoring state from 'Connecting' to 'Operational'
Recv:  T0:27 /0 T1:26 /0 B:24 /0
Recv:  T0:27 /0 T1:26 /0 B:24 /0
Recv: ok
Send: N1 M20*16
Unexpected error while writing to serial port: AttributeError: ''NoneType' object has no attribute 'rfind'' @ comm.py:_doSendWithoutChecksum:1752
Changing monitoring state from 'Operational' to 'Error: AttributeError: ''NoneType' object has no attribute 'rfind'' @ comm.py:_doSendWithoutChecksum:1752'
Connection closed, closing down monitor

foosel, 2016-03-14:
That appears to be a bug with the GPX plugin rather than OctoPrint itself. please report this in the [plugin's bug tracker](https://github.com/markwal/OctoPrint-GPX/issues).

closed 2016-03-14 by foosel

markwal, 2016-03-14:
That's funny.  We were just talking about line numbers and checksums and boom, there appears to be a bug in my code to remove the checksum.  :-)

gddeen, 2016-03-14:
2016-03-14 10:55:51,542 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.7
2016-03-14 10:55:51,547 - octoprint.server.util.sockjs - WARNING - Could not send message to client 192.168.1.7: 'utf8' codec can't decode byte 0x8b in position 53: invalid start byte

Is that this???