OctoPrint/OctoPrint #1281: [Request] Link temperature, serial speed and other  specifics to the printer profile.

ntoff, 2016-03-17:
It would be nice to have all the printer specific things linked to a particular profile. For example my Mendel90 uses 250000 baud while the other thing I have uses 115200. Both of them also have different print temperatures for ABS (245 vs 235) and the heated bed.

As of now, as far as I can tell, the only things the profile changes are the size of the print surface, whether it has a bed or not, and a few speed and acceleration related things. That leaves serial baud rate, and the hot end/bed temperatures as separate "global" settings.

Am I wrong? 

Note: I am not trying to connect two printers simultaneously, My DIY printer is down so I just swapped the pi over to the Mendel to print out some new parts.

foosel, 2016-03-18:
No, you are not wrong, and that's been something I've been thinking about for a while now too. The way I see it, there are quite a lot of settings defined globally now that would make more sense to attach to a printer profile or maybe a new entity (maybe a device profile which holds a printer model plus serial settings plus temperature stuff plus...).

The following things are currently defined globally but actually do apply to the printer connection and hence would make more sense as a per-connection/printer-setting:

  * serial port
  * serial baudrate
  * timeouts and polling rates
  * hello command
  * long running commands
  * commands that always require a checksum
  * wait for start on connect
  * always assume sd card is present
  * send a checksum with every command
  * ignore consecutive resend requests for the same line
  * support TargetExtr%n/TargetBed target temperature format
  * disable detection of external heatups
  * support resend requests without ok (that one's new)
  * the temperatures
  * the terminal filters (partially)
  * the GCODE scripts (partially)

The problem is that the printer profiles as they are currently implemented also fuse together physical machine specifics (name, color) and printer model specifics (build volume, axes speeds, heated bed or not etc). So it would probably make sense to decouple these things to allow better composition.

Basically something like a "Printer" definition that consists of:

  * meta data such as a name and associated color
  * connection settings (port + baudrate)
  * firmware settings (all the protocol specific flags mentioned above)
  * printer profile (build volume, axes speeds etc)
  * temperatures
  * specific gcode scripts for various events
  * specific terminal filters

And have that associated with the connection instead of the printer profile now.

Problem is that this is a quite disruptive change - there are plugins out there that depend on the status quo, so a way would need to be found to ensure backwards compatibility here. Also, existing configurations would need to be ported over without losing data, possibly requesting input from the user (the new wizard subsystem on the `devel` branch might help with that though).

gddeen, 2016-03-18:
I've made my carriage to snap in/out extruders. I'm still thinking on this since some characteristics of the printer do change. Perhaps the new UUID is also a printer value stored in above.

ntoff, 2016-03-18:
In that case I probably don't need it that badly. I mean who really swaps their pi between multiple printers?

Could I have multiple config.yaml's? So say I could have one for the Mendel and one for the Frankenprinter and if I'm using the Mendel I could log in via putty and "mv config_mendel.yaml config.yaml" restart octoprint and have it just have all the Mendel stuff in there. Are there other places where octoprint stores settings and things? I'm not too particularly bothered about the statistics of how many prints printed or how many failures there were.

Seems like that might be easier (if it works) in the short term.