OctoPrint/OctoPrint #1284: [Request] Support for the latest version of CuraEngine

CutEdge, 2016-03-18:
I've searched all over to see if this has already been brought up, and I don't believe it has. Please correct me if I'm wrong.

I think it would be beneficial to have the latest version of CuraEngine working with OctoPrint. As of now, I am running the "legacy" version. It works fairly well for me, but I wish I could take advantage of some of the newer settings. Namely, the ability to do XY compensation. (For instance, adding .5mm to the diameter of holes, without having to design it into the part). I think that feature was introduced in Cura 15.06 (https://github.com/daid/Cura/issues/1194).

foosel, 2016-03-23:
There is [preview of a plugin with 15.10 support here](https://github.com/OctoPrint/OctoPrint-CuraEngine) (currently still needs OctoPrint's `devel` branch!). 

You need a completely separate plugin since the CuraEngine versions are so completely different with regards to slicing profiles and slicer parameters between up-to-15.04 and anything after this that it's pretty much impossible to "just" extend the existing plugin (and I know that because @nicanor-romero and me tried to get just that to work a couple months back).

And I'll probably extract the current Cura plugin into an external plugin in a later release as well, the current bundling was a bad decision.

Later Cura versions (officially unreleased btw, 15.06 was retconned as beta and the official downloads still only offer 15.04.x versions) also made it way more difficult to even export existing profiles, making a custom profile editor necessary. Which becomes extra tricky thanks to the hierarchical slicing profile format which can depend on a chain of files which might exist in an existing Cura installation but not automatically be also available on a user's OctoPrint setup. 

So, anyone desperately needing support for the unreleased CuraEngine versions is welcome to contribute to the plugin above. I cannot say much about its state, since to be honest, I've currently lost track a bit of that and can't take a look right now.

CutEdge, 2016-03-23:
In the mean time, does anyone know if there's a workaround to use XY Compensation within CuraEngine 15.04?

Renha, 2016-07-26:
I was looking into Octo's Cura code, and it's so complicated, I have no idea even on how to make a "copy" of it in OctoPrint, i.e. to have 2 folders with code with 2 puncts in interface, settings, etc. to experiment with.

Meanwhile, working with new curaengine is simple, it just have different file format, while being still easily editable by hand (you just write needed values as default ones). Machine files could be easily be taken from Cura installation.

Additional plus is that format is chained, so you could import files from Cura and don't change them, instead create new settings file and import them and override values in it. You could even include your own files, so current settings hell should be gone:

I have some amount of settings files in OctoPrint right now, like abs-thin-nosupport, abs-thinnest-brim, pla-thick-touching, and much more combinations. Now imagine I've found out that for example if first layer would be 0.21 mm instead of 0.2 mm, I would get much better adhesion. Now i need to edit all of these files to fix that! And in my current Octo installation, afaik, I couldn't even download them from OctoPrint - I need to find PC or phone where each file was created, make sure it's final version, edit and upload to Octo. And if file couldn't be found, recreate it. Repeat for each file.

With new CuraEngine machine profile I could have base file and some files that could be chained with it, and first layer thickness could be changed for all profiles at once.

foosel, 2016-07-26:
> I have no idea even on how to make a "copy" of it in OctoPrint, i.e. to have 2 folders with code with 2 puncts in interface, settings, etc. to experiment with.

You literally copy the plugin folder, give it another name and also replace the occurrences of the "cura" name in the plugin's ``__init__.py`` with said other name.

> Additional plus is that format is chained

Actually, that's a big minus, since it makes it completely impossible to reuse the old storage implementation and forced us to completely abandon any hope for a slicer profile abstraction layer for good, which sucks hard since it makes it even more complicated to provide a unified slicing dialog that allows overriding slicing parameters _on the go_ for every possible plugged in slicer. Yes, it is still possible by having slicer plugins declare their supported editable parameters + names and letting them have take care of proper parameterization, but boy does this make the implementation more complicated than just offering up a standardized interface.

But again, feel free to experiment with the above linked plugin and improve on it.

Klipi, 2016-11-11:
@foosel, Is there anything new on the slicer side or is the info here still up to date? I'd like to try and make a plugin for CuraEngine 2.3. Would you recommend using the 15.10 plugin as a starting point?

foosel, 2016-11-11:
@Klipi nothing new here, everything of the above is still valid. I guess the 15.10 plugin is also still the best starting point, but I can't be 100% sure, I haven't looked at CuraEngine 2.x+ at all yet.

alancnet, 2017-05-20:
I wonder if there is any news? I desperately need the XY offset feature of Cura 2

Renha, 2017-05-31:
@alancnet just in case you have delta: check arms length, wrong length leads to wrong XY scale.

If not, sorry :)

Also, i'm with you on waiting for news. I had actually tried to write plugin, but code structure of existing plugins wasn't clear enough for me (it's that I am not clever enough, not that code was bad), so I failed.

foosel, 2017-05-31:
No news from my side, and this is also not going to change in the foreseeable future. Consider this up for grabs.

DongerZone, 2018-12-16:
Thinking about grabbing this. Might make this my holiday break project. If successful would we be able to incorporate this into the default build instead of Cura 15.04?

foosel, 2018-12-17:
I don't think that @guysoft would object to that. Would be great though if you could do it as a standalone plugin. I'll remove the 15.04 plugin from OctoPrint sooner or later (not remove as in delete but extract into a standalone plugin that needs to be installed instead of coming bundled).

I'm not sure if it helps, but [there's this ancient attempt at getting the first Cura version with the API to work as a slicer plugin](https://github.com/OctoPrint/OctoPrint-CuraEngine). It's not viable as is, but there might be some bits and pieces in there that could be useful. No promises though.

guysoft, 2018-12-17:
If you manage to do that I am for it. 
Related:
https://github.com/guysoft/OctoPi/issues/465

Also there were talks on adding slic3r as a plugin:
https://github.com/guysoft/OctoPi/issues/444

DongerZone, 2018-12-17:
Consider it grabbed, I started working on it last night.
I might even make it an all around compatible slicer plugin (ie works for Slic3r, Cura15.x, Cura2.x, etc?)

foosel, 2018-12-18:
Marked it as grabbed :)

alancnet, 2018-12-22:
So glad to hear this is making headway!

jneilliii, 2020-05-12:
Haven't seen anything in this ticket that @DongerZone made any headway. I have recently compiled the latest 4.6.1 CuraEngine on a Pi3 and although it was a little fiddly I was able to get it to work and slice an stl file into gcode. Has anyone else picked up this work or have any updates?

@guysoft Guy, I'm not sure what you might need to implement the build in your OctoPi image, but if you could provide some details I might be able to give you the necessary info.  The below are the commands that I went through to get the dependencies installed and compile the executable for CuraEngine. Not sure if this works in Python 2 but will attempt, so might make sense to plan on adding this to a future release of OctoPrint 2.0.0. There also could be some unecesssary steps trimmed out.

```
sudo apt-get install libtool dh-autoreconf cmake git python3-setuptools python3-sip-dev
cd ~
wget https://github.com/protocolbuffers/protobuf/releases/download/v3.11.4/protobuf-python-3.11.4.tar.gz
tar zxf protobuf-python-3.11.4.tar.gz
cd protobuf-3.11.4
./autogen.sh
./configure
make
sudo make install
cd python
python3 setup.py build
sudo python3 setup.py install
cd ~
git clone https://github.com/Ultimaker/libArcus.git
cd libArcus
mkdir build
cd build
cmake ..
make
sudo make install
cd ~
git clone https://github.com/Ultimaker/CuraEngine.git
cd CuraEngine
mkdir build
cd build
cmake ..
make
```

guysoft, 2020-05-12:
@jneilliii We basically need this with an ``if`` switch inside .
This is the old cura code, I guess you/me could just update it:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/start_chroot_script#L104

foosel, 2022-01-18:
The CuraEngine plugin is no longer part of OctoPrint and slicing is not really in the scope anymore either.

closed 2022-01-18 by foosel

guysoft, 2022-01-18:
That being said, if someone wants to become the "person who seriously maintains slicers on OctoPi" I would consider it.