OctoPrint/OctoPrint #1285: "Load and print" button broke in latest update

giantpune, 2016-03-18:
I have octopi running with the gpx plugin. Just now I received an update for "OctoPrint: 1.2.10". After that update, the octopi restarted. Now I can list the gpx files on my sd card as before, but clicking the little "Load and print" button does nothing. I have to press the buttons on the front of my printer and browse to the file to print it like some sort of caveman.

I posted this issue in the octopi tracker and they said to post it here.  https://github.com/guysoft/OctoPi/issues/213

There is nothing in the octoprint.log file indicating what the error is.
```
2016-03-18 21:56:05,132 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.2.10...
2016-03-18 21:56:30,671 - octoprint.plugins.softwareupdate - INFO - Update of octoprint to 1.2.10 successful!
2016-03-18 21:56:30,892 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-03-18 21:56:30,900 - octoprint.plugins.softwareupdate - INFO - Restarting...
2016-03-18 21:57:06,161 - octoprint.server - INFO - Starting OctoPrint 1.2.10 (master branch)
2016-03-18 21:57:06,221 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-03-18 21:57:06,471 - octoprint.plugin.core - INFO - Found 6 plugin(s) providing 6 mixin implementations, 7 hook handlers
2016-03-18 21:57:06,660 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-03-18 21:57:06,667 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-03-18 21:57:06,718 - octoprint.plugins.softwareupdate - INFO - Version cache was created for another version of OctoPrint, not using it
2016-03-18 21:57:08,739 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-03-18 21:57:08,744 - octoprint.plugin.core - INFO - Initialized 6 plugin implementation(s)
2016-03-18 21:57:08,751 - octoprint.plugin.core - INFO - 6 plugin(s) registered with the system:
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/discovery
|  GPX (2.4.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/virtual_printer
2016-03-18 21:57:08,753 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-03-18 21:57:08,758 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-03-18 21:57:08,763 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-03-18 21:57:08,789 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-03-18 21:57:09,894 - octoprint.plugins.GPX - INFO - Connecting through x3g.
2016-03-18 21:57:09,898 - octoprint.plugins.GPX - INFO - GPXPrinter created, port: /dev/ttyACM0, baudrate: 115200
2016-03-18 21:57:11,931 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-03-18 21:57:11,954 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-03-18 21:57:11,959 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-03-18 21:57:12,218 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2016-03-18 21:57:12,294 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-03-18 21:57:12,425 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.2.35/'}
2016-03-18 21:57:12,441 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.101
2016-03-18 21:57:13,513 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

2016-03-18 21:57:14,547 - octoprint.plugins.GPX - INFO - gpx.connect succeeded
2016-03-18 21:57:16,039 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning

2016-03-18 21:57:19,253 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
2016-03-18 21:57:19,320 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-03-18 21:57:27,116 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'en', 'path': '/', 'base_url': 'http://127.0.0.1/'}
2016-03-18 21:57:29,634 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'en', 'path': '/', 'base_url': 'http://192.168.2.35/'}
2016-03-18 22:25:58,673 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.2.101
2016-03-18 22:26:01,023 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.101
2016-03-18 22:29:20,457 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.2.101
2016-03-18 22:29:22,679 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.101
2016-03-18 22:29:49,069 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.2.101
2016-03-18 22:29:51,216 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.101
2016-03-18 22:49:44,817 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.2.101
2016-03-18 22:49:46,421 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.2.101
```

giantpune, 2016-03-18:
serial tab and log are in this gist.  https://gist.github.com/giantpune/5414e810070582900b7e

foosel, 2016-03-21:
Load and print button works just fine in 1.2.10 with stock OctoPrint, so my guess is it could be an issue with 1.2.10 vs the current GPX plugin.

@markwal can you comment on this?

markwal, 2016-03-21:
I agree.  If there's a bug here it is in the GPX plugin.  Bug tracker for that is over here: https://github.com/markwal/OctoPrint-GPX.

Sorry about the run around @giantpune, you're using software that is made up of layers written and maintained by different authors.  The GPX plugin has all the stuff specific to your printer's firmware (Sailfish) that talks a different protocol than everything else (x3g).  It is running on OctoPrint which is, in turn, running on an OS image (in your case OctoPi, but can be run on other OS's like Windows and Mac).

markwal, 2016-03-21:
Shoot.  I can't try and repro now because I'm in a middle of a print trying to reproduce the serial communication error, but my guess now is that there is something wrong with the ".x3g" extension hook.

markwal, 2016-03-21:
I'm able to reproduce this issue.  Files with .x3g extension can't be selected.

markwal, 2016-03-21:
I believe the new version of the GPX plugin (2.4.2) fixes this issue.

closed 2016-03-21 by markwal

giantpune, 2016-03-22:
myth confirmed.  the latest GPX plugin fixed it for me.