OctoPrint/OctoPrint #1286: [BUG] [Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:1918

IAmOrion, 2016-03-19:
Print fails at random points with the following error:

Unexpected error while writing to serial port: IOError: '[Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:1918
Changing monitoring state from 'Printing' to 'Error: IOError: '[Errno 5] Input/output error' @ comm.py:_do_send_without_checksum:1918'
Closing down send loop
Connection closed, closing down monitor

With version 1.2.9 I **_NEVER_** had this issue, since updating to 1.2.10 yesterday it has happened one every single print, failing at what seems like random times.

foosel, 2016-03-21:
There's an error when OctoPrint tries to send data to the printer. There's nothing I can do about that. The fact that it only happens with the sailfish based firmware and not the i3 hints at there possibly being an issue with the GPX plugin, causing this problem. But it's unclear. In any case, it's not a bug in OctoPrint, if OctoPrint can't send data to the printer, it can't send data to the printer.

@markwal can you say something about that? Anything in 1.2.10 that might have shaken up the gpx plugin?

IAmOrion, 2016-03-21:
Is there a method to downgrade to 1.2.9 ? I found something you wrote on a different post dated 13 Feb however it didn't work, it was if I had just reinstalled the latest, even though I used the 1.2.9 branch.

foosel, 2016-03-21:
Your first line is wrong, it shouldn't be `~oprint/bin/activate` but `~/oprint/bin/activate`. Yes, there's a difference ;)

Downgrading for OctoPi in a nutshell:

```
source ~/oprint/bin/activate
git reset --hard <target version>
python setup.py install
sudo service octoprint restart
```

**Important**: No (!!!) `sudo` for the `python setup.py install` step. You broke your setup that way, [you need to fix it **first**](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cannot-updateswitch-branches-i-get-something-about-permissions):

```
sudo chown -R pi.pi ~/OctoPrint
```

(Yes, **here** you indeed need sudo)

----

So, in your specific case:

```
sudo chown -R pi.pi ~/OctoPrint
cd ~/OctoPrint
source ~/oprint/bin/activate
git reset --hard 1.2.9
python setup.py install
sudo service octoprint restart
```

IAmOrion, 2016-03-21:
"Sugar plum fairies" as they say instead of an expletive.... woods for the trees and all that, can't believe I missed the ~/ ... rather annoyed with my self now!

Thank you so much.  Hopefully @markwal can pitch in at some point regarding the GPX plugin.  Every single print fails with the same error, so if it's not Octoprint, I guess the GPX Plugin would be the next suspect.  9 times out 10 it literally fails after a single wall/outline.  1 out of 10 it finishes 2 layers of the brim.  Always the same error though regardless.

Thanks again @foosel

markwal, 2016-03-21:
I mostly test with the devel branch and haven't seen anything like this.  I'll checkout 1.2.10 and see.

That I/O Error gets thrown by the GPX plugin when it receives ESIOWRITE, ESIOREAD, ESIOFRAME or ESIOCRC from the GPX layer.  There are some circumstances where it will retry, but at the moment most failures are one and done.  It's always been that way since that was the way the GPX module I started with was written by whpthomas.  I probably should make it a little more resilient and retry most failures.

However, I don't think there's any change in OctoPrint that could make this more or less likely except for just timing.  Perhaps 1.2.10 got faster enough that it makes it more likely we're able to get ahead of the bot in a way that it doesn't reply properly.

@IAmOrion, it could be that your serial connection is picking up noise from your motor wires.  Switching it out for a different cable or even just re-positioning it may change your results.  Did going back to 1.2.9 fix it?

IAmOrion, 2016-03-21:
@markwal I use a shielded USB cable that extends directly from the rear, in a straight line to the wall where my Pi is wall mounted.

So far, (I've only tested 1 small print) it has worked, it printed.  I will retry a longer print and report back later tonight.

markwal, 2016-03-22:
I can't reproduce this issue in either OctoPrint 1.2.10 or on devel branch.  Many hours of printing today with 1.2.10 with no problem.  We should probably move this to the GPX plugin buglist: https://github.com/markwal/OctoPrint-GPX.  @IAmOrion, could you include one of the models that reproduce the issue?  Perhaps it is a command sequence that causes the serial glitch.  Though it sounded like a random spot every time.

IAmOrion, 2016-03-22:
@markwal I agree, this most like needs to be moved to the GPX bug list.  I will upload some GCode files shortly and then transfer this to the GPX issues page.  However, yes, every time was random.  A print started, completed just 1 outline, then failed with the above error.  I restarted the same print, this time it completed 2 layers of the raft before it failed with the same error.  There was a point it got to 65% and I thought it might finally make it but no, it failed with same error, so I don't think it's any specific command as thus far it has never happened at the same point

markwal, 2016-03-22:
Ok. Thanks for copying it and the extra info.

closed 2016-03-22 by markwal