OctoPrint/OctoPrint #1287: Silent crash of 1.2.10

thhart, 2016-03-19:
I have a rather long print (10 MB - 8 hours). Yesterday night OctoPrint simply died for unknown reason and the printer stopped in the print keeping heating. The Raspberry itself was reachable 
Unfortunately octoprint.log contains nothing than my last connections. Also /var/log/messages is not showing essential information. Is there anything else I should check or send to help diagnose?

foosel, 2016-03-21:
That's a tough one, I don't see anyone that might be responsible for this in the log either.

You wouldn't have checked with `ps -ef | grep -i octoprint` if the process was still running when you noticed it not responding anymore?

It sounds like a full out crash, but it should actually log all exceptions to the log file, so it sounds like it crashed so hard it didn't even get around to this anymore. I've not seen this happen in quite a while (and when it was due to the whole pi going down, not just the server).

Can you reproduce it? Has it happened again?

thhart, 2016-04-23:
Unfortunately I could not reproduce nor did it happen again. I will close the issue but will announce if happen again.

closed 2016-04-23 by thhart