OctoPrint/OctoPrint #1288: Issues with printing on Ras. PI 2... Help?

willh2much, 2016-03-21:
Please read the "guidelines for contributing" that are linked ^-- just
up there. Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions). Mark requests with
a [Request] prefix in the title please. Fully fill out the bug reporting
template for bug reports.

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Thank you!

----

#### What were you doing?

I was printing my gcode file through octoprint.

#### What did you expect to happen?

I expected the printer to print the object flawlessly (like it did with Octoprint on windows)

#### What happened instead?

The print will go for a certain amount of time (could be seconds or minutes) then it will stop right where it is. About a minute later, it will start back printing... like nothing happened.

#### Branch & Commit or Version of OctoPrint

Octopi ver. 0.12
OctoPrint ver. 1.2.2

#### Printer model & used firmware incl. version

This a custom made printer, Marlin Firmware, running on GeeeTech printrboard clone (same as printrboard Rev. D)

#### Browser and Version of Browser, Operating System running Browser

Latest version of Chrome on Mac El Capitan (probably unrelated)

#### Link to octoprint.log

Cannot provide my log because i do not have access to my PC right now but the log from issue #568 is nearly identical, because his issue is very similar. his problem didn't seem to be solved though. 
https://gist.github.com/BJClark/43556873ac5ded7e3ad2

...Basically its a completely normal log...

#### Link to contents of terminal tab or serial.log

Completely normal terminal then this shows up:

(Date and time stuff) octoprint.comm.protocol - WARNING - Printer requested line (Some random Line #(different every time)) but no sufficient history is available, can't resend

#### Link to contents of Javascript console in the browser

Irrelevant (probably) will produce if necessary.

#### Screenshot(s) showing the problem:

N/A... Not important...

I have read the FAQ.

markwal, 2016-03-21:
More than likely you experienced a communication timeout and OctoPrint forced a line to restart things.  The default timeout is set to 30 seconds if I recall correctly.  The serial log would be very helpful for this issue.  It's possible that the python serial library difference between the pi and windows has play here, but more likely is that your pi and the USB cord are in a different place than when you were using Windows.

willh2much, 2016-03-21:
Alright, I have linked the serial log (up to the point of one of the errors) below now that i have access to my computer:

http://pastebin.com/ECn2xjs7

foosel, 2016-03-22:
You are running into a communication timeout. Your printer simply stops responding to OctoPrint while printing. That is not caused by OctoPrint, that is caused by the connection between your Pi and your printer or maybe your printer itself:

```
2016-03-21 18:51:54,558 - SERIAL - DEBUG - Send: N2367 M105*23
2016-03-21 18:51:54,586 - SERIAL - DEBUG - Recv: ok T:229.9 /230.0 B:0.0 /0.0 T0:229.9 /230.0 @:57 B@:0
2016-03-21 18:51:54,597 - SERIAL - DEBUG - Send: N2368 G1 X92.119 Y93.886 E30.38661*106
2016-03-21 18:52:24,634 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-03-21 18:52:24,636 - SERIAL - DEBUG - Send: N2369 M105*25
2016-03-21 18:52:54,650 - SERIAL - DEBUG - Communication timeout during printing, forcing a line
2016-03-21 18:52:54,653 - SERIAL - DEBUG - Send: N2370 M105*17
2016-03-21 18:52:54,665 - SERIAL - DEBUG - Recv: Error:Line Number is not Last Line Number+1, Last Line: 2367
2016-03-21 18:52:54,668 - SERIAL - DEBUG - Recv: Resend: 2368
2016-03-21 18:52:54,671 - SERIAL - DEBUG - Recv: ok
2016-03-21 18:52:54,671 - SERIAL - DEBUG - Send: N2368 G1 X92.119 Y93.886 E30.38661*106
2016-03-21 18:52:54,678 - SERIAL - DEBUG - Recv: ok
```

Note how it times out after sending lines 2368 and 2369. Then your printer wakes up again after a minute of no replies at all, claims it never got 2368 and wants it resent. OctoPrint obliges. Then something else happens:

```
2016-03-21 18:52:54,680 - SERIAL - DEBUG - Send: N2369 M105*25
2016-03-21 18:52:54,691 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,701 - SERIAL - DEBUG - Send: N2370 M105*17
2016-03-21 18:52:54,712 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,723 - SERIAL - DEBUG - Send: N2371 M105*16
2016-03-21 18:52:54,734 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,744 - SERIAL - DEBUG - Send: N2372 M105*19
2016-03-21 18:52:54,755 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,766 - SERIAL - DEBUG - Send: N2373 M105*18
2016-03-21 18:52:54,777 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,787 - SERIAL - DEBUG - Send: N2374 M105*21
2016-03-21 18:52:54,799 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,810 - SERIAL - DEBUG - Send: N2375 M105*20
2016-03-21 18:52:54,820 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,831 - SERIAL - DEBUG - Send: N2376 M105*23
2016-03-21 18:52:54,842 - SERIAL - DEBUG - Recv: ok T:230.0 /230.0 B:0.0 /0.0 T0:230.0 /230.0 @:51 B@:0
2016-03-21 18:52:54,852 - SERIAL - DEBUG - Send: N2377 M105*22
[...]
```

A whole cascade of temperature queries - and this has been fixed in 1.2.10. You are running a version of OctoPrint nearly a year old. It is probably screaming at you asking you to upgrade. Please upgrade.

So, in summary, not a bug. We can continue this discussion here, but I'm marking this as closed, since it's a) not OctoPrint's fault and b) even if it was, not reproduced against 1.2.10.

closed 2016-03-22 by foosel

willh2much, 2016-03-22:
I forgot to mention, the last time i tried to update though octoprint panel
my pi was bricked by it (or at least octoprint was) and i had to re image

willh2much, 2016-03-22:
Resolved by following guide on enabling software update for versions below 1.2.5

z3dev, 2016-03-23:
Just one comment... USB cables can provide static charges between PC and 3D printer. Be sure that the power is sourced from the same power bar, and don't play with other electrical devices during printing. And make sure that the 3D printer is grounded properly internally, and externally. Good luck!

gddeen, 2016-03-23:
If it behaves in a timed fashion, maybe the print bed is turning on and driving voltages low. Try printing without heating the bed. Look to see if anything is dimmed when printing. Is it a RPI? If so make sure it has adequate power.  Your printer board is capable of going to sleep and continuing when power resumes (in microseconds). So, maybe your print bed needs replacement or a new power supply.

Just a guess. (I did think it reported brownouts over the serial communications port though.) But Octoprint can be configured to ignore and keep going...

willh2much, 2016-03-23:
No heated bed, power supply has about 40w overhead power when extruder is
heating and steppers are active. I solved the problem though... I had a USB
extension going from my raspi to the micro usb that went into the printer
board. I unplugged the extension and used JUST the micro usb and? WORKED
FLAWLESSLY *FACEPALM*