OctoPrint/OctoPrint #1290: [Request] Have the "Time Left" value be more precise

david-vo, 2016-03-22:
It'd be nice if the time left estimate could go down to the minutes rather than the nearest hour. Approximate print time (estimate) and elapsed print time both display to the second accuracy but time left (for print) is will say "1 hour" even if 1:20 is left. It's inconsistent and not very helpful.

markwal, 2016-03-22:
Elapsed print time should be fairly accurate.

Approximate print time and time remaining are fairly approximate and "1 hour" communicates the correct amount of precision.  To be consistent, the approximate print time should be shaved of a few sigfigs rather than the other way around.  But consistency isn't everything.  I kinda like it the way it is.

KC703, 2016-03-22:
Maybe we just need to add "Approx" to the Print Time Left field...

david-vo, 2016-03-22:
I've been watching a print I am doing via Octoprint and am wondering how it's even calculating Print Time Left. It seems like anything under an hour will give it in minutes but anything over will round to the hour. So 1:20 hours = 1 hour. But 0.5 hours displays 30 minutes. Is the thinking here that 1 hour and 1 hour , 20 minutes is roughly the same thing? What is the cut off point? 1:30 = 2 hours?

markwal, 2016-03-22:
Yes, the closer you get to the end of the print, the more accurate the estimate can be because it can use the print time of the first part to calculate the rest.  And yes anything over an hour, I would say is at least +/- 30 min. in my experience because it's hard to tell what the effect of your firmware's acceleration will be on the actual time consumed by the moves (always slower than specified in the gcode).  As the print proceeds, the algorithm switches over to portion of file consumed.

david-vo, 2016-03-22:
Ok that makes sense. I guess it's fine as is. I usually prefer more info than less but it's not a huge deal and it does make sense that it gets more accurate as you near completion. At least this spawned an informative, to me, discussion.

foosel, 2016-03-23:
I've sunken weeks into trying to more accurately predict a non-linear process through linear interpolation. Not going to happen.

closed 2016-03-23 by foosel