OctoPrint/OctoPrint #1292: Cura plugin settings translation changes token expansion in start/end gcode

markwal, 2016-03-23:
#### What were you doing?

My start gcode in my cura profile includes the following:
```
M104 S{print_temperature} T0
M104 S{print_temperature2} T1
```
I uploaded that profile to OctoPrint and use it to slice with the CuraEngine plugin

#### What did you expect to happen?
I expected the output gcode to have something like:
```
M104 S200 T0
M104 S190 T1
```

#### What happened instead?
```
M104 S200 T0
M104 S?print_temperature2? T1
```

#### Branch & Commit or Version of OctoPrint

devel c433e32e

#### Link to octoprint.log

http://hastebin.com/ejepewuhuj

I have read the FAQ.

markwal, 2016-03-23:
print_temperature2 gets translated into a list entry on import, but isn't reversed for token expansion.

foosel, 2016-03-23:
Just for the record, you are whitelisted in the bot, so you don't need to follow the template fully, especially not if you are just creating self-assigned bugs anyhow :)

Would be great if you could do fixes on `fix/...` branches derived from `maintenance` and then merge those into both `maintenance` and `devel`, I've found that to keep the history way more clean than cherry-picking fix commits all over the place (PRs being an exception here since those seem to consist more of feature additions than bug fixes).

(And yes, I know, I really need to get around to write that kind of stuff down in some sort of development guide, I've really not been very forth coming there...)

markwal, 2016-03-23:
Ah.  OK.  Will do.

I'm not sure how far to go with the old Cura plugin since Cura 2.1 might be right around the corner and then we'll want to switch to the other plugin for active development probably.

For multiple extruders, there are a couple of other problems that make this not quite "upload your profile, slice, get same results".  One is that Cura desktop only uses the start2.gcode and end2.gcode if you also tell it to use the other extruder for something (like support).  The plugin always uses it if you have two extruders.

Then I was thinking, it wouldn't be that hard to make a plugin for modifying profiles on the fly, but then again, not sure if it would be worth the time since we might be moving on at some point here.

foosel, 2017-01-26:
This actually was fixed in 1.3.1 while I was tackling #1708.

closed 2017-01-26 by foosel

markwal, 2017-01-26:
Ah. Nifty. :-)