OctoPrint/OctoPrint #1295: Default mjpeg url to current ip address

hcker2000, 2016-03-23:
It would be nice if OctoPrint would assume that mjpg streamer was running on the same host unless otherwise specified.

This way if your using a dynamic ip you dont need to worry about updating configs for mjpg streamer every time you ip shifts.

For example if octoprint is being accessed at 192.168.0.33 then thats the ip it should output if mjpg streamer is configured unless you change the stream urls.

markwal, 2016-03-23:
The default configuration of octopi (different project, but has mjpg-streamer already set up) does just that.  It sets the stream url to be server relative: /webcam/?action=stream and the snapshot url to be the loopback address that doesn't change with dynamic ip: http://127.0.0.1/webcam/?action=snapshot

Should work no matter what ip address you get.

Perhaps your request is to have OctoPrint default to the same urls as octopi assuming that if a user sets up mjpg-streamer, they'll also set up haproxy and put the mjpg-streamer redirect off of the same url (webcam)?  Unfortunately, I'm not sure if that is the common case though.

foosel, 2016-03-24:
It isn't a common case. OctoPrint is completely platform agnostic. It shouldn't assume that a webcam will be served by mjpg-streamer, on the same host it's located on itself, with a reverse proxy rewriting connections to port 80 and `/webcam` path prefix. That's not it's place.

On Windows, the recommended server to use is Yawcam, which has completely separate URLs. People might want to use existing old phones with ip camera apps as webcam, which is perfectly possible. It's a configuration option people need to fill out (and which they get filled out by OctoPi).

closed 2016-03-24 by foosel