OctoPrint/OctoPrint #1296: SD card not printing with latest marlin downloaded in Jan 2016

mhk3000, 2016-03-24:
#### What were you doing?

I am using version 1.2.10 (master branch) octoprint. I can raise temperature by setting it manually from the temperature tab, but if I am runnng a gcode file from the SD card, nothing happens. 

Gcode files get uploaded locally, but not on SD card. And therefore, files do not get downloaded from SD card. Furthermore, the gcode file which I have uploaded locally onto the Pi, the Print command is working. 

I think there is some communication problem with the SD card. 
#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint
Version 1.2.10 (master branch)

#### Printer model & used firmware incl. version
I am using one of the latest version of Marlin. We have built our own 3D Printer
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome 50. Mac OSX 10.11
[If applicable, always include if unsure.]

#### Link to octoprint.log
https://www.dropbox.com/s/7txn6beu2n5ip9a/octoprint.log?dl=0

#### Link to contents of terminal tab or serial.log
https://www.dropbox.com/s/apb0mjr0wh1may0/serial_log.pdf?dl=0

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser
because of its detrimental effects to the end user's experience. For more help, check https://xhr.spec.whatwg.org/.
packed_app.js:9015 Starting dependency resolution...
packed_app.js:9072 ... dependency resolution done
packed_app.js:9326 Going to bind 21 view models...
packed_app.js:9394 ... binding done
[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:
https://www.dropbox.com/s/dayh8tsuqyku9ba/Screen%20Shot%202016-03-24%20at%204.00.14%20PM.png?dl=0

I already have "Send a checksum with every command” disabled. 
I have read the FAQ.

jneilliii, 2016-03-24:
This is not octoprint related,  but there's a new define in configuration.h in the latest rc branch of marlin you have to uncomment. It's SD card support or sdsupport, can't remember which.

foosel, 2016-03-24:
```
Send: N1 M23 /pillar~1.gco*28
Recv: open failed, File: pillar~1.gc.
```

Your Marlin version is cutting off the final character of the file name. [That's a firmware bug that's actually in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cannot-select-files-on-the-printers-sd-card). Have the firmware fixed.

closed 2016-03-24 by foosel