OctoPrint/OctoPrint #1300: Cannot see local files list

AmedeeBulle, 2016-03-28:
#### What were you doing?

This is rather stupid from my side, but if you enable 'Show only files on SD-Card', then disable SD-card support, you can't see your local files anymore.
This is per say not a big issue, but in a shared environment when one leaves Octoprint in that state, other users don't understand why they don't see their files...

To reproduce:

1. Enable SD-Card support (if not active)
1. Restart Octoprint (looks to me a restart is needed to activate / deactivate that setting)
1. In the Files box, select 'Show only files on SD-Card'
1. Disable SD-Card support
1. Restart Octoprint

You don't see any files, and there is no option in the menu to show them (Unless you re-enable SD-Card support)

#### What did you expect to happen?

When SD-Card support is disabled, re-enable local files

#### What happened instead?

Empty files box ;) (Even if you upload a new file)

#### Branch & Commit or Version of OctoPrint

Version: 1.2.10.post0.dev0+g692166f (master branch)

#### Printer model & used firmware incl. version

N/A (Ultimaker Original)

#### Browser and Version of Browser, Operating System running Browser

N/A (Firefox 45.0.1)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

markwal, 2016-03-28:
In the devel branch when you do this, the file list is empty (as reported here), however, there is still an option to show the local files in the menu.  Is that sufficient?  In which case this is already fixed for a future release.

markwal, 2016-03-28:
@foosel I fixed this in fix/noSdCardNoSdFilter branch and merged into maintenance.  However, merging into devel brings 1a4c06d6 with it since the fix branch was based on maintenance and I wasn't sure you wanted that merged yet.  I could cherry-pick or do the fix branch at a229cbfd to get the fix into devel.  Or, of course, merge maintenance.  Sorry about that.  Which would you prefer?

foosel, 2016-03-29:
@markwal Merging anything from `maintenance` into `devel` that's not already there is totally fine - it usually is just caused by me not having yet merged any fixes or improvements up after a release.

Changes to the versioneer lookup table thingy are especially unproblematic since they relate to the branch, so merging anything to `devel` that doesn't change the `devel` line in the file is pretty much without consequence :)

markwal, 2016-03-29:
183ca32 is in maintenance and devel, so this fix should show up in the next maintenance release (1.2.11).

closed 2016-03-29 by markwal