OctoPrint/OctoPrint #1301: Unable to activate manual extrusion

Jack-2025, 2016-03-29:
#### What were you doing?
Adding a new filament and trying to extrude from Control >> Tool (E) >> Select Tool "Hotend" >> click "Extrude" button.

#### What did you expect to happen?
"Hotend" is activated and the printer extrudes the specified mm after hitting the "Extrude" button.

#### What happened instead?
When clicking "Hotend", the combo box reverts to "Select tool..." and there is no extrusion.

#### Branch & Commit or Version of OctoPrint
Version: 1.2.10 (master branch)

#### Printer model & used firmware incl. version
MakerGear M2

#### Browser and Version of Browser, Operating System running Browser
Chrome

#### Link to octoprint.log
http://pastebin.com/kn7FPYxZ

foosel, 2016-03-29:
> Link to contents of terminal tab or serial.log
>
> NA

...

shaiss, 2016-03-29:
I have the same issue where I have hotend in the drop down but cannot select it.  here's my terminal log http://pastebin.com/ZCtzrwdX.

I'm working on getting the serial.log.  I have a wanhao i3 repetier fw based

markwal, 2016-03-29:
By the way, the "Select Tool" drop down is just a bunch of command buttons that send a "select tool" gcode to your printer "T0" for example.  It doesn't show status because it doesn't really know what state your printer is in.  You just use that button to tell your printer how to interpret subsequent commands.  So the part where it still says "select tool" is expected.

The part that isn't expected is that extrusion doesn't work.  And for that, we need the terminal tab or the serial.log.  (Most likely your printer wants the tool selection gcode to be different which you'll be able to customize, but you can also not click it if you only have one extruder like shaiss.)

shaiss, 2016-03-29:
ok, by some magic it seems to work and selecting does nothing as expected.  probably as I only have extruder.  But the gui should then not allow me to select anything and just say Ext 1

Jack-2025, 2016-03-29:
I've added the terminal tab contents

markwal, 2016-03-29:
OK, looks like OctoPrint is behaving as expected.  Here is the relevant bit of your terminal I/O:
```
Send: T0
Recv: echo:Active Extruder: 0
Recv: ok
Send: G91
Recv: ok
Send: G1 E5 F300
Recv: echo: cold extrusion prevented
Recv: ok
Send: G90
Recv: ok
Send: M105
Recv: ok T:16.9 /0.0 B:17.5 /0.0 T0:16.9 /0.0 @:0 B@:0
Send: T0
Recv: echo:Active Extruder: 0
Recv: ok
Send: G91
Recv: ok
Send: G1 E5 F300
Recv: echo: cold extrusion prevented
Recv: ok
Send: G90
Recv: ok
```
You can see that the select tool button is sending a T0 which is switching to your right extruder and the bot likes that.  And then the extrude button is saying hey, give me 5mm of filament (G1 E5 F300).  And the bot says "hey, your hotend is too cold to do that".  You might want to use the Temperature tab to preheat to an extrusion temperature before telling it to extrude.

markwal, 2016-03-29:
The key thing to realize is that all the buttons on the "control" tab are "send some gcode" to the printer.   There isn't any info coming the other way (except through your camera, but OctoPrint isn't looking at that, it just passes it along to you).

closed 2016-03-29 by markwal

Jack-2025, 2016-03-29:
Thanks all for the replies.
I'd have expected some kind of error popup from within the Control panel indicating that the command failed and the reason.  If the "Recv: echo: cold extrusion prevented" is a message particular to my printer and there's no way to tell that the gcode execution actually failed, yeah I guess debugging the problem through the terminal or camera is the only way to go.

markwal, 2016-03-29:
Unfortunately some firmwares respond to gcode out of order so there's no way to know which response goes to which gcode in the OctoPrint code to show you the response to your button.  The information that OctoPrint does show you in the UI like temperature panel is all parsed, expected responses.  Recv: echo: cold extrusion prevented is unfortunately fairly firmware specific.

AntDX316, 2020-05-26:
Whoever said that you have to have the temperature up first gave 100% the right answer.