OctoPrint/OctoPrint #1302: Octoprint Crash

droidmunkey, 2016-03-29:
Running on an Original Raspberry Pi

2016-03-29 19:27:50,199 - octoprint.server - CRITICAL - Now that is embarrassing... Something really really went wrong here. Please report this including the stacktrace below in OctoPrint's bugtracker. Thanks!
2016-03-29 19:27:50,206 - octoprint.server - ERROR - Stacktrace follows:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/server/__init__.py", line 519, in run
    ioloop.start()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv6l.egg/tornado/ioloop.py", line 830, in start
    fd, events = self._events.popitem()
KeyError: 'popitem(): dictionary is empty'
2016-03-29 19:27:50,330 - octoprint.server - INFO - Shutting down...
2016-03-29 19:27:51,280 - octoprint.server - INFO - Goodbye!

markwal, 2016-03-29:
@droidmunkey, what version of OctoPrint are you using?  Also, what version of tornado?  If you're running off of an octopi image you can use the following command line to get the versions of the python dependencies you are using: `cd ~/OctoPrint ; ~/oprint/bin/pip freeze`

droidmunkey, 2016-03-29:
pi@octopi ~ $ cd ~/OctoPrint ; ~/oprint/bin/pip freeze
Babel==1.3
Flask==0.10.1
Flask-Assets==0.10
Flask-Babel==0.9
Flask-Login==0.2.2
Flask-Principal==0.3.5
Jinja2==2.7.3
MarkupSafe==0.23
OctoPrint==1.2.10
OctoPrint-ActionTriggerPlugin==0.1
OctoPrint-EmailNotifier==0.1.1
OctoPrint-M3DFio==0.30.1
OctoPrint-PrintHistory==0.9
OctoPrint-RequestSpinner==0.1.1
OctoPrint-Stats==1.0.0
OctoPrint-TitleStatus==0.0.4
PyYAML==3.10
RPi.GPIO==0.5.11
STL-Viewer==0.3.1
Unidecode==0.04.19
Werkzeug==0.8.3
XlsxWriter==0.8.4
argh==0.26.1
argparse==1.2.1
awesome-slugify==1.6.5
backports.ssl-match-hostname==3.4.0.2
blinker==1.3
certifi==2015.04.28
distribute==0.6.24
itsdangerous==0.24
keyring==8.5.1
netaddr==0.7.17
netifaces==0.10.0
numpy==1.6.2
pathtools==0.1.2
picamera==1.10
pifacecommon==4.1.2
pifacedigitalio==3.0.4
pkginfo==1.2.1
psutil==3.2.1
pyasn1==0.1.8
pybonjour==1.1.1
pygame==1.9.1release
pygobject==3.8.2
pylru==1.0.9
pyserial==2.7
pytz==2015.4
regex==2016.03.02
requests==2.7.0
rsa==3.2
sarge==0.1.4
semantic-version==2.4.2
sockjs-tornado==1.0.1
speaklater==1.3
tornado==4.0.1
virtualenv==1.7.1.2
watchdog==0.8.3
webassets==0.10.1
wsgiref==0.1.2
yagmail==0.4.129

markwal, 2016-03-29:
Hmmm...  the tornado code looks a little suspect to my eye, but I'm used to pre-emptive multi-threading.  With the GIL, perhaps it isn't so much: https://github.com/tornadoweb/tornado/blob/master/tornado/ioloop.py#L874-L880  The comment there is interesting though.

This thread also is interesting: https://groups.google.com/forum/#!topic/python-tornado/wPz7SaqtaQk

droidmunkey, 2016-03-29:
I saw that stuff-  according to them it's a bug w/ multi-threading and something calling functions directly from the wrong thread (instead of using event queues to pass messages back to the main thread to call the function) - but if that was true why isn't this a common problem?

I have horrible problems with the web browser disconnecting from OctoPrint, i constantly have to refresh the page to get the 'sync with job progress' feature of gcode viewer to work. I assume it's because i'm using an original raspberry pi which is too slow (very first ones, from the actual kickstarter campaign for the pi!).   loading the web interface can take 10+ seconds for everything to show up and top shows 100% load on the pi

My only guess is it's a race condition related to slow hardware on a single core machine?

I don't have any steps to reproduce, So i guess I will try again tonight and hope it doesn't happen.

I was in the middle of printing a case for a rasp pi 3 when this happened! argh, so close to moving to a faster system....  Maybe i will just hook it up w/out a case temporarily :-)

Thanks!

foosel, 2016-04-14:
Forgot to mark that ticket as not incomplete and the bot did it's job, reopened, sorry for that.

foosel, 2018-10-29:
So maybe (but really just maybe) @gdombiak just ironed out a possible reason for this in #2850 or rather #2858, as that definitely caused code to be written outside of tornado's event loop if the timing was just right. I still haven't been able to reproduce this particular issue as described here even once though, so I can't verify that the fix for #2850 also fixes it or more digging might be needed.

gdombiak, 2018-10-29:
Same as Gina. Never experienced this kind of error so cannot tell if #2850 helps or not.

foosel, 2019-11-05:
Cautiously assuming this has been fixed since i haven't heard anything to the contrary in ages.

closed 2019-11-05 by foosel