OctoPrint/OctoPrint #1303: Temperature polling doesn't work on 1.2.10

lucacoelho, 2016-04-01:
#### What were you doing?

A fresh install of the 1.2.10 tag, connected to a Wanhao Duplicator i3 printer.  Check the temperature diagram.  The temperature only changes sporadically (after issuing other commands), but the polling and reporting doesn't work.

#### What did you expect to happen?

The temperature graph should be updated regularly with the new temps.

#### What happened instead?

The graph and the temperature values are stuck on a stale value.

#### Branch & Commit or Version of OctoPrint

commit 692166f067329cd3d6fdc84389e0dd76184c5e0c (tag: 1.2.10)

#### Printer model & used firmware incl. version

Wanhao Duplicator i3 with the firmware installed in the factory (based on Repetier 0.91? -- I can confirm soon, if needed).

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 49.0.2623.87 (64-bit), running on Debian testing.

#### Link to octoprint.log

I'll provide logs soon, but I could see in the logs that the periodic M105 commands were not sent.

#### Link to contents of terminal tab or serial.log

See above.

#### Link to contents of Javascript console in the browser

Not available right now, I can provide on request.

#### Screenshot(s) showing the problem:

Irrelevant.

I have read the FAQ.

lucacoelho, 2016-04-02:
Change the title and the description to point out that 1.2.9 doesn't work either on my setup.

lucacoelho, 2016-04-02:
Closed accidentally.

lucacoelho, 2016-04-02:
Here are the logs.

[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/200746/octoprint.log.txt)
[serial.log.txt](https://github.com/foosel/OctoPrint/files/200747/serial.log.txt)

foosel, 2016-04-02:
Try checking all options in Settings > Features that have a little
"Repetier" next to them. Repetier firmware sends temperature responses in a
way that make them look like external heatup events in other firmware, so
the detection for that needs to be disabled.

lucacoelho, 2016-04-02:
Yes, that did the trick! Thanks.

Is there any way this could be enabled automatically when a Repetier firmware is detected?

markwal, 2016-04-11:
Maybe firmware detection could work, but considering the snowflake situation we have with firmware's and versions and forks, it may not be worth the trouble.

closed 2016-04-11 by markwal

lucacoelho, 2016-04-11:
Yeah, I know very well how hard it is to detect features by looking at firmware versions. ;) I was thinking more about something like "You appear to be using a Repetier-based firmware, would you like to active Repetier-specific options?"

But yeah, probably not worth the trouble.