OctoPrint/OctoPrint #1304: Print Time Left Inaccurate

jneilliii, 2016-04-03:
#### What were you doing?
Printing a part.

#### What did you expect to happen?
For the print time left to display the approx-total print time minus actual print time.

#### What happened instead?
Print time left was almost 30 minutes longer than the expected calculation.

#### Branch & Commit or Version of OctoPrint
1.2.10 (master branch

#### Printer model & used firmware incl. version
Marlin 1.1.RCBugFix branch

#### Browser and Version of Browser, Operating System running Browser
Google Chrome Version 49.0.2623.110 m running on Windows 10 64 Bit.

#### Link to octoprint.log
n/a

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
https://gist.github.com/jneilliii/0c35e92356e3561fbb174a8833badea4

#### Screenshot(s) showing the problem:
Sorry didn't catch it in time before the print was finished.

I have read the FAQ.

jneilliii, 2016-04-03:
I originally had just searched for open issues on this and went back and realized that this has been discussed before in closed tickets as something that may not be possible.  Just wondering why the print time left isn't just a simple calculation of approx-total print time minus actual print time? Wouldn't that be more appropriate than trying to interpolate?

markwal, 2016-04-03:
No.  Actually, just doing the subtraction would be easier, but it would be a lot less useful.  It's trying to use what it knows about how long it actually to print so far, to predict how long the time remaining will be.  The main problem now is that improving this takes a significant amount of effort and the gains are small, so it's not a very fruitful place to focus effort.  It may be someday, but today there is much lower hanging fruit for improving OctoPrint.

closed 2016-04-03 by markwal