OctoPrint/OctoPrint #1305: Feedback Command not working

luke321, 2016-04-04:
Hello,

I have a raspberry 2 with OctoPi installed and I want to use the feedback_command to read a certain eeprom value.

I tried adding the following example to the config.yaml

- command: M114
      default: ''
      name: Get Offset
      regex: "X:([0-9.-]+)Y:([0-9.-]+)Z:([0-9.-]+)E:([0-9.-]+)"
      template: "Position: X={0}, Y={1}, Z={2}, E={3}"
      type: feedback_command

The button appears and M114 is send to the printer but no Output is shown next too the button.
Also the lines in the config.yaml change after restarting Octoprint, I suppose this is the problem:

- command: M114
      default: ''
      name: Get Offset
      regex: X:([0-9.-]+)Y:([0-9.-]+)Z:([0-9.-]+)E:([0-9.-]+)
      template: 'Position: X={0}, Y={1}, Z={2}, E={3}'
      type: feedback_command

does anyone know how to get this work?

luke321, 2016-04-06:
#### What were you doing?

I tried adding the following example to the config.yaml

command: M114 default: '' name: Get Offset 
regex: "X:([0-9.-]+)Y:([0-9.-]+)Z:([0-9.-]+)E:([0-9.-]+)" 
template: "Position: X={0}, Y={1}, Z={2}, E={3}" type: feedback_command
The button appears and M114 is send to the printer but no Output is shown next too the button.
Also the lines in the config.yaml change after restarting Octoprint, I suppose this is the problem:

command: M114 default: '' 
name: Get Offset 
regex: X:([0-9.-]+)Y:([0-9.-]+)Z:([0-9.-]+)E:([0-9.-]+) 
template: 'Position: X={0}, Y={1}, Z={2}, E={3}' 
type: feedback_command

#### What did you expect to happen?

That "Position: X={0}, Y={1}, Z={2}, E={3}" with actual position shows up next to my command button.

#### What happened instead?

Nothing. The Gcode Command M114 is send to the printer but the reply is not parsed and shown in the gui, altough the printer sends back the correct position.

#### Branch & Commit or Version of OctoPrint

1.2.10 (master branch)

#### Printer model & used firmware incl. version

Makergear M2 with Repetier Firmware 0.92

#### Browser and Version of Browser, Operating System running Browser

Windows 10, Google Chrome, should not matter.

#### Link to octoprint.log

[octoprint.log](http://pastebin.com/N5KhT95e)

#### Link to contents of terminal tab or serial.log

[Terminal content after pressing the Button](http://pastebin.com/BMwQ2wre)

#### Link to contents of Javascript console in the browser
[JaveScript Console content](
http://pastebin.com/3kcSmCEk)

Salandora, 2016-04-06:
Could you please upload your config.yaml.

luke321, 2016-04-06:
Here you go:

[config.yaml](http://pastebin.com/husqUSRP)

markwal, 2016-04-06:
Doesn't your regex need to allow spaces between the parameters?  It doesn't seem like it would match a line with spaces unless I'm misreading it.

luke321, 2016-04-06:
Hey,

I used the example from [here](https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls), what would you exactly change to use spaces, I have no clue about regex.

luke321, 2016-04-06:
Thx for your input markwal,

I checked the regex with [this](https://regex101.com/) regex tester and changed it to:

`X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)`

to match the output of the printer for M114 which is:

`X:0.00 Y:0.00 Z:0.00 E:0.00`

I found the new regex [here](http://docs.octoprint.org/en/master/features/custom_controls.html).

But I still get not
Also the regex and the template are in between quotation marks.
When I try to add quotation marks " " or inverted commas ' ' the always get removed from the regex in my config.yaml after a restart.

When I remove the values from the Template like this

`template: 'Position:'`

the template shows up next to the Button.

So either the regex does not get machted or it does not get parsed to the template.

markwal, 2016-04-06:
Good to hear it.

closed 2016-04-06 by markwal