OctoPrint/OctoPrint #1309: Help needed: Virtual printer problem

sanchosk, 2016-04-11:
Hi, all.
I am working on a small modification to Octoprint - to move from Canvas to SVG.
However, I run into 2 strange issues with the Virtual printer.
First of all, it's lightning fast - when I try to print with it, the printer does not adhere to the settings of mm/minute - the calculated speed of print is within the range of 3 minutes for 1.5 hours G-code.
Second, after I put in a parameter rxBuffer and lowered it from 64 to 16, I receive communication errors like this:
```Unexpected error while writing to serial port: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:1922
Changing monitoring state from 'Printing' to 'Error: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:1922'
Closing down send loop
Connection closed, closing down monitor```

My config.yaml part for the virtual printer follows:
```devel:
  virtualPrinter:
    commandBuffer: 4
    enabled: true
    forceChecksum: false
    includeCurrentToolInTemps: true
    movementSpeed:
      e: 300
      x: 3000
      y: 3000
      z: 200
    numExtruders: 1
    rxBuffer: 16
```

sanchosk, 2016-04-11:
#### What were you doing?

 I am working on a small modification to Octoprint - to move from Canvas to SVG.
However, I run into 2 strange issues with the Virtual printer.
First of all, it's lightning fast - when I try to print with it, the printer does not adhere to the settings of mm/minute - the calculated speed of print is within the range of 3 minutes for 1.5 hours G-code.
Second, after I put in a parameter rxBuffer and lowered it from 64 to 16, I receive communication errors like this:
```Unexpected error while writing to serial port: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:1922 Changing monitoring state from 'Printing' to 'Error: SerialTimeoutException: '' @ comm.py:_do_send_without_checksum:1922' Closing down send loop Connection closed, closing down monitor```

My config.yaml part for the virtual printer follows:
```  virtualPrinter:
    commandBuffer: 4
    enabled: true
    forceChecksum: false
    includeCurrentToolInTemps: true
    movementSpeed:
      e: 300
      x: 3000
      y: 3000
      z: 200
    numExtruders: 1
    rxBuffer: 16
```

#### What did you expect to happen?

I expected the print to last for ~1 hour and print as slowly as my real life printer does.

#### What happened instead?

It printed too fast and produced communication errors on virtual serial port.

#### Branch & Commit or Version of OctoPrint

    Version: 1.2.10.post0.dev0+g692166f (master branch) 

#### Printer model & used firmware incl. version

Virtual printer from the master branch repo.

#### Browser and Version of Browser, Operating System running Browser
Firefox 45.0 on Ubuntu 64bit
Linux elsa 4.2.0-34-generic #39-Ubuntu SMP Thu Mar 10 22:13:01 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
Distributor ID:	Ubuntu
Description:	Ubuntu 15.10
Release:	15.10
Codename:	wily

#### Link to octoprint.log

[http://pastebin.com/raw/KCjz4efn](url)

#### Link to contents of terminal tab or serial.log

[http://pastebin.com/raw/jzYNYREn](url)

#### Link to contents of Javascript console in the browser

N/A (due to development, I had a lot of side messages in console and I lost the content :( )

#### Screenshot(s) showing the problem:

Not needed, see the log.

Yes, I have read the FAQ.

markwal, 2016-04-11:
The rxBuffer just shrinks the number of characters in the processing buffer, it won't slow your virtual print down.  It's used for testing buffer overflow handling which you may have found, but 16 is a pretty small buffer.

To slow down the print, I think you want to slow how fast the virtualPrinter processes the incoming (_processIncoming in virtual.py) since that simulates what the printer is doing.  There is a setting "devel, virtualPrinter, supportF" that if set to true, simulates the duration by sleeping inside of performMove. (in the devel branch anyway)

sanchosk, 2016-04-12:
Thanks for the hint.
As I don't want to get another branch (I'm afraid I'll break more things than I'll fix by that) I simply placed a generic 0.5s wait into performMove. Seems (based on the code) there is already a facility to wait based on the command, I just don't know python and from the code I did not found the way to populate the dictionary for _sleep_after option :)
However, I can continue to work now, so I am closing the issue. Thanks!

closed 2016-04-12 by sanchosk