OctoPrint/OctoPrint #1311: [Request] M240 Timelapse trigger

Thump2010, 2016-04-12:
#### What were you doing?

I want to have the timelapse trigger on the M240 Conveyor event
I tried implementing the change myself but couldn't get it to work, I had it showing up as an option but the option would never save.
To get it to work for now I changed the on z change event from Z_Change to CONVEYOR and put M240 in my slicers layer code and see if it worksbut that is not an option everyone can do.
I see how easy this could be implemented, I just wish I knew python enough to do it and contribute it.

#### Branch & Commit or Version of OctoPrint

1.3.0.dev973+ga2672f3.dirty (devel branch)

#### Printer model & used firmware incl. version

Homebuilt rc5 of latest marlin

#### Browser and Version of Browser, Operating System running Browser

opera

I have read the FAQ.
yes

MoonshineSG, 2016-06-02:
whats the difference ? if the M240 is in the layer change, wouldn't this be the same as zchange timelapse ?

Thump2010, 2016-06-02:
No, on z change if you have z hop or anything z related then it makes the timelapse not so nice but a m240 is on when the layer has actually changed, no software guessing etc. 
It makes for much nicer videos.

MoonshineSG, 2016-06-03:
Apparently, there are plans to change the whole snapshots module to allow custom implementations via plugins.

If you can't wait, have a look at https://github.com/MoonshineSG/OctoPrint/commit/506643e883bdef80e2e8b90c29d7ab54742334fc 

I had a custom M888 gcode implemented in Marlin firmware use as trigger for snapshots. You could modify it to suit your own needs. 

The new event gets triggered by https://github.com/MoonshineSG/OctoPrint-Plugins/blob/7403e6113ad163b28bb5751f800265e17dd94682/snapshot.py

NOTE: I no longer use this, so the links are to files from history.