OctoPrint/OctoPrint #1312: OctoPi changes config.yaml rearranges and Modifies events

WVUTampaAlum, 2016-04-15:
I am trying to setup octopi to send me a picture of the object when the print is completed.  I running octopi and Version: 1.2.10 (master branch).  I have followed the guide on (https://github.com/foosel/OctoPrint/wiki/Cookbook:-Useful-Eventhooks) .  I have tried 4 times and cannot get it to send me an email but running the same 'command' in terminal I get the email with pic attached.  What I have found from this is that octoprint is moving the section I enter (I put it at the bottom) and completely change the events.

As a test I entered the following from the wiki.  I then restarted octoprint server made 0 changes and viewed config.yaml.  It was then completely changed around.

Entered in Events section/ from wiki:
events:
  enabled: True
  subscriptions:
  - event: Disconnected
    command: python ~/growl.py -t mygrowlserver -d "Lost connection to printer" -a OctoPrint -i http://raspi/Octoprint_logo.png
    type: system
    enabled: false
  - event: PrintStarted
    command: python ~/growl.py -t mygrowlserver -d "Starting {file}" -a OctoPrint -i http://raspi/Octoprint_logo.png
    type: system
  - event: PrintDone
    command: python ~/growl.py -t mygrowlserver -d "Completed {file}" -a OctoPrint -i http://raspi/Octoprint_logo.png
    type: system

Restart octoprint and config.yaml is changed to (its moved events and now added dashes for command line):

events:
  enabled: true
  subscriptions:
  - command: python ~/growl.py -t mygrowlserver -d "Lost connection to printer" -a
      OctoPrint -i http://raspi/Octoprint_logo.png
    enabled: false
    event: Disconnected
    type: system
  - command: python ~/growl.py -t mygrowlserver -d "Starting {file}" -a OctoPrint
      -i http://raspi/Octoprint_logo.png
    event: PrintStarted
    type: system
  - command: python ~/growl.py -t mygrowlserver -d "Completed {file}" -a OctoPrint
      -i http://raspi/Octoprint_logo.png
    event: PrintDone
    type: system

markwal, 2016-04-15:
Yes. OctoPrint reads config.yaml into a dictionary.  If it decides to write any settings, it rewrites the entire file by re-exporting the settings dictionary as yaml.

Typically, the dictionaries get sorted by key.

markwal, 2016-04-15:
Github markdown also changed the snippets you posted here.  You can put triple quotes around code to avoid that difficulty.  I've edited your post so that the snippets are easier to read.

In any case, this rewriting of the config.yaml is expected and in the example you gave, will have no effect on the run time (since the internal dictionaries for each event will work like they did before).  It's merely putting each dictionary in that event array in key order (command, event, type) and since those keys are looked up by key, it's the same either way around.

closed 2016-04-15 by markwal

WVUTampaAlum, 2016-04-15:
The real problem I face is that it appears octoprint does not execute events in config.yaml presumably because it rewrites everything.  Are you saying the rewritten code is correct?  Is there any way to debug config.yaml/events and what not?

Here is what I want.
events:
  subscriptions:
  - event: PrintDone
  command: curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of {file} finished" /tmp/printDone.jpg <your mail address>
  type: system

That code doesn't appear to run.  There is not jpg and I never get any email.  I can run in terminal and code executes fine.
command: curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of {file} finished" /tmp/printDone.jpg <your mail address>

markwal, 2016-04-15:
Two things that might help:

1. You have to get the indentation precisely correct.  Yaml insists, like python, that precisely the right number of spaces and/or tabs match all of the other items at the same level.  Notice in your earlier example, the event item dictionary keys were lined up with each other, not with the "-" marking the array entry.
2. You need to get OctoPrint to read it after you've add your entry.  You do this by restarting OctoPrint: "sudo service octoprint restart".  If you hit "Save" on settings before the restart, you'r new entry will be overwritten.  A very cautious approach would be to stop octoprint, edit, and then restart:

```
sudo service octoprint stop
nano ~/.octoprint/config.yaml
sudo service octoprint restart
```

```
events:
  subscriptions:
  - event: PrintDone
    command: curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of {file} finished" /tmp/printDone.jpg <your mail address>
    type: system
```

Also, the command runs in the context of the pi user, so the /tmp folder must be writable by that user.  I assume that you've got that part working since you said you can run from the teminal (without sudo).

markwal, 2016-04-15:
Also, I notice your second post didn't have "enabled: True" in it.  Since we're just talking about snippets it's easy to conclude that it is already there, but just in case, I thought I'd mention it.