OctoPrint/OctoPrint #1313: [Request] Option to render timelapse later instead of immediately

mwhuss, 2016-04-15:
Feature Benefits:

Sometimes prints fail or I want to start another print immediately which is not recommended when rendering a timelapse. I'd like the ability to go to "Unrendered Timelapses" and start the render myself instead of it happening immediately after a print finishes.

philphilphil, 2016-04-18:
I would like this as well, my rpi1 is soo slow and often it's not even worth rendering when the print fails.
Maybe with a new setting "Autostart Timelapse renders" so nothing changes for users who like the auto-start of the render.

If foosel likes this idea i would also be able to implement it :).

foosel, 2016-04-18:
An "Autostart timelapse renders" checkbox on the timelapse tab sounds great, feel free to go ahead with that @philphilphil! Just a word of a warning, I'm currently a bit behind with reviewing code changes due to having my hands full with the [funding situation](https://plus.google.com/u/0/+OctoprintOrg/posts/5m53jf6cdme), so I don't want to promise to be able to immediately merge it when you send a PR. Sorry in advance for that.

philphilphil, 2016-04-18:
@foosel no worries :). I think i'll need some days to finish and test it correctly anyway.

mwhuss, 2016-04-18:
@philphilphil let me know if you need any help testing it

fraternl, 2016-09-29:
A lot of days have passed since April 18th ;-)
I just created a ticket that's somewhat related:
https://github.com/foosel/OctoPrint/issues/1526

philphilphil, 2016-09-30:
@fraternl i kind of forgot about this ticket.. I have already some changes from back in April on my computer, i will review them and finish it, still an important feature i think :)

fraternl, 2016-10-04:
@philphilphil  do you consider https://github.com/foosel/OctoPrint/issues/1526 as well in your changes?

fiveangle, 2017-10-02:
I can see this isn't getting much traction, but for consideration of posterity, a preference to "Automatically delete Cancelled print time lapses" (low hanging fruit) as well as "Render timelapse when not printing" options would be ideal.  For the when-not-printing scenario, the script could subscribe to the log file (or port to syslog if not already) to monitor when the job is finished, or cancelled, then automatically kick off the render(s).  There would have to be some coordination trigger mechanism so that if OP started a new print job, it would notify the render script to quit or pause render/resume "monitor" mode, waiting for completion, then restart the job.

Oninaig, 2021-11-15:
Any update(s) on this?

grantx2016, 2022-03-18:
I would like this also.

Avrution, 2022-03-26:
Same.  Really not sure why there isn't an option to not render on cancel.

Avrution, 2022-08-02:
> I can see this isn't getting much traction, but for consideration of posterity, a preference to "Automatically delete Cancelled print time lapses" (low hanging fruit) as well as "Render timelapse when not printing" options would be ideal. For the when-not-printing scenario, the script could subscribe to the log file (or port to syslog if not already) to monitor when the job is finished, or cancelled, then automatically kick off the render(s). There would have to be some coordination trigger mechanism so that if OP started a new print job, it would notify the render script to quit or pause render/resume "monitor" mode, waiting for completion, then restart the job after.

This

foosel, 2022-08-02:
> Really not sure why there isn't an option to not render on cancel.

Because no one has sent a pull request to do so, and I have my hands full with other stuff 🙂 But I've now marked this as up-for-grabs in the hopes that someone will look into it.

WisdomCode, 2024-04-23:
@fiveangle Thanks for summing up possible routes and options for the issue, that was helpful for evaluating possible implementations. Let me in the same vein sum up my design rationale so it can be built upon and evaluated.

I think the gold standard would be to simply pause rendering whenever a print is running. This would basically prevent any problems arising from CPU usage while printing without any user intervention or understanding. However, in its current design, I dont think its possible: We are using ffmpeg, and I don't think there is a way to pause and resume such a command. One could fragment the rendering process entirely and concat the result, but that would be a very complex rewrite with unknown results (not to mention possibly even more user options that may be confusing), so I wouldnt expect that to happen unfortunately, despite its obvious usefulness. I would love to be proven wrong here.

Deleting failed prints automatically is actually an option that may be added easily, but has been left out for now. The reason is that, depending on the reason why it was cancelled, one may want to check want went on (or wrong), so there is an advantage at keeping the raw files for a certain time, which is already the default. On the other hand, a few GB of raw data is barely worth the consideration for most, and cleaning it up is rather easy, be it manually or automated. So the target audience for such an option I would consider pretty small (maybe pi users with <32GB SD Card and a constantly in use Printer). Consider those unrendered files as a "recycle bin" for restoring footage just in case.

So whats left is the actual trigger of when to render. Currently it is rendered always after a print has been terminated. I added the option to render either only if a print terminated successfully or never render them automatically. In both cases, the rendering can still be triggered manually or via API calls at any given time later.

foosel, 2024-06-12:
This was implemented by @WisdomCode 🎉 and will be part of 1.11.0

closed 2025-04-22 by foosel