OctoPrint/OctoPrint #1316: [REQUEST] Delete old timelapses automatically

thhart, 2016-04-23:
#### What were you doing
I am recording timelapse video just for the records, just in case something did go wrong it is great to have these. However the list of videos grows very long and it is annoying to delete these manually.

For OctoPi (OS for running OctoPrint on Raspberry) I can instantiate a cron job but IMHO it should be integrated into OctoPrint itself:
`find /home/pi/.octoprint/timelapse -mtime +7 -delete`

#### What did you expect to happen?
Old aged timelapse should be removed automatically after a configurable time.

#### Branch & Commit or Version of OctoPrint
1.2.10 and above

#### Printer model & used firmware incl. version
TAZ 5

#### Browser and Version of Browser, Operating System running Browser
n/a

#### Link to octoprint.log
n/a

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s) showing the problem:
n/a

jamsessionein, 2018-04-20:
Would find this pretty useful.

Alwinator, 2020-12-29:
I would also need this feature! Are there any news?

jneilliii, 2020-12-29:
You could set this up yourself as a cron job on the pi using the command listed.

```
find /home/pi/.octoprint/timelapse -mtime +7 -delete
```

foosel, 2021-01-25:
Would also be a prime candidate for a plugin

jneilliii, 2022-01-23:
I've created a plugin that will delete timelapses older than a configured amount of days. It will purge on startup of OctoPrint and on completion of a new timelapse rendering. 

https://github.com/jneilliii/OctoPrint-TimelapsePurger

jneilliii, 2022-01-23:
another alternative would be to use Event Manager with the previously listed command `find /home/pi/.octoprint/timelapse -mtime +7 -delete` on any event you choose.

closed 2022-02-06 by cp2004