OctoPrint/OctoPrint #1318: Base temperature missing from GUI with 1.2.10

normanclanger, 2016-04-26:
Please read the "guidelines for contributing" that are linked ^-- just
up there. Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions). Mark requests with
a [Request] prefix in the title please. Fully fill out the bug reporting
template for bug reports.

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Thank you!

----

#### What were you doing?

I upgraded to 1.2.10

#### What did you expect to happen?

#### What happened instead?

The GUI only shows the Hotend temperature on the graph - the bed is not displayed, nor is the temperature displayed under the graph.  The temperature is being reported via the terminal.

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface.]

Version 1.2.10 Master

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

Velleman K8200

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

Windows 7, Chrome & IE

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser

[On gist.github.com
[image]
 or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2016-04-26:
Does your printer profile has the "heated bed" check box checked? If not, there's your problem. 

Also see [this gif](http://i.imgur.com/wp1j9bs.gif)

normanclanger, 2016-04-26:
Yep that was my problem - wow what customer service!  I'd just rolled back to 1.2.7 and got the temperature back, but am now at 1.2.10 with that little box ticked and all is fine.  And you've added some nice 'please wait' messages as the server starts for the impatient amongst us!  Thanks Tim.

closed 2016-04-26 by normanclanger