OctoPrint/OctoPrint #1319: [Request] Create a *.deb package

Knetic, 2016-04-27:
While struggling to get [Octopi to work|https://github.com/guysoft/OctoPi/issues/226], I found myself desperately wishing that Octoprint just had a \*.deb file that I could use to install onto my Pi. Making these packages is dead simple, and makes installation/uninstallation/maintenance of any product into a rather trivial matter requiring zero user knowledge of how the product is actually put together. But Octoprint doesn't seem to make any packages, the only realistic way to install it seems to be building from source using a bunch of random Python-specific tools, or by _flashing a whole new OS with Octopi_.

Just adding a build step that invokes `checkinstall` or `fpm` to make a \*.deb would completely remove any operational pain in getting this project to work.

It's possible that I'll get frustrated enough to do this myself in a PR, but I want to leave this issue here in case this is already on a roadmap, or in case someone else is working on it, or if the idea itself is anathema to the goals of the project.

markwal, 2016-04-27:
You may be interested in: #845 Debian Packaging

Knetic, 2016-04-27:
Yep, that looks like what I'm after! Looks like it ended up in [OctoPrint-Packaging](https://github.com/croadfeldt/OctoPrint-Packaging). Seems like I can pull Octoprint master, copy over that packaging repo, and wind up with a \*.deb.

It seemed like that was positively received, but I don't see any deb files in the releases list on github, nor anywhere on the site. Am I looking in the wrong place? Is there a PPA somewhere that I missed?