OctoPrint/OctoPrint #1322: new version of file -> load and print buttons disabled.

rewolff, 2016-05-01:
I design stuff, so regularly I need to load-and-print a new version of a design I just printed. Anyway, to prevent mixups I used to delete the old version before uploading the new version. 

Just now I got (over) confident and uploaded the new version while octoprint was still in the "print finished" state. The new version is then drawn in black (not printed), but the load and print buttons don't work. I had to select another design, then reselect the one I wanted, and could then "load-and-print". (in the meanwhile the web interface had loaded the gcode for the OTHER design and it didn't switch automatically back to the active one, probably because I clicked too fast. 

I'm running the version I pulled from git with tag "preparing for 1.2.10".

foosel, 2016-05-01:
The "load and print" button is disabled because your fresh file is already selected. Have you tried what happens if you just hit the big blue "print" button?

rewolff, 2016-05-01:
Ehhmmmm. Maybe I'm supposed to feel a bit stupid now... :-) 

OK! How do I know for sure that the new file is selected? I need better feedback on that... (I overlooked whatever feedback there was)

So, suppose I'm printing object A, then upload object B, it will keep on printing A. When it finishes hitting print again will give me another A, But when it's DONE printing A and I upload B and hit print, I get an object B, right? That's a bit inconsistent. On the other hand, I guess that this is a good tradeoff in usability for "normal use".

foosel, 2016-05-01:
As long as A and B have the same name X, it won't allow you to upload it while you are already printing it. Reason being that you would basically rip the file it's currently sending line by line from underneath it otherwise. 

If you replace a file with another one named the same when it is already selected for printing but not printing (= status is operational, file name is in status section) it will update the metadata (including size and date) and the gcode preview.

rewolff, 2016-05-01:
And that replacing the active file only happens when the current file has the same name as what I'm uploading? I just uploaded a different file, and it remained unselected. 
The metadata that updated was "subtle": I changed a minor thing: The object essentially stayed the same. soooo.... 
Feature request: Make it plainly visible when an object has been printed to the end (hopefully successfully). e.g. by a green logo (or background?) in the meta-info plane. A fresh file gets started with

foosel, 2016-05-01:
It doesn't automatically select a newly uploaded file (unless you have the autoselect plugin installed), but if you have currently selected a file named X and you upload a new file also named X it will replace that currently selected file X with the new one and still  have it selected. 

Stuff you printed until the end does get a green font in the file list already. Might make sense to do that in the state pane as well.

rewolff, 2016-05-02:
I've been thinking about this while printing some more over the weekend. (My printer is shaping up. In the beginning I had to go and watch the beginning and cancel if things were not going as planned, now I just start the job and drop by once or twice during printing :-) )

**Suggestion**: 
I think the STATE window should have a file-state as well. The state could be:
- Just uploaded, analyzing
- Just uploaded, analyzed. 
- uploaded a while ago. 
- printing 
- cancelled print
- printed
- printed twice

closed 2016-05-16 by GitIssueBot

rewolff, 2016-11-12:
I had just uploaded and selected a print when I realized: Ooops I left "fill" at 100%, not necessary for this print. (I've had prints where fill=100% improved printing speed.). 
I changed the fill setting, saved the GCODE, uploaded it, and hit print.... 
Turns out it is still printing the 100% fill version. On this print that takes a lot of time and a lot of unnecessary filament....