OctoPrint/OctoPrint #1324: Daemonized OctoPrint does not clean up pidfile anymore

kevans91, 2016-05-04:
#### What were you doing?

```
octoprint --daemon start --pid /tmp/octoprint.pid
kill -TERM `cat /tmp/octoprint.pid`
ls -l /tmp/octoprint.pid
```

/tmp/octoprint.pid persists -- this was previously addressed in issue #711 with PR #712, but it has since broken. The following SIGTERM registration clobbers this handler: https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/__init__.py#L511-L516 -- so the daemonize'd SIGTERM does not get called, and the pidfile does not get cleaned up.

#### What did you expect to happen?

pidfile to get cleaned up

#### What happened instead?

pidfile did not get cleaned up

#### Branch & Commit or Version of OctoPrint

Verified on:

master & maintenance: e79703b
devel: 1b1148d

#### Link to octoprint.log

https://gist.github.com/kevans91/b11ea167ec1fd9d4c4159400608abcd2

I have read the FAQ.

foosel, 2016-05-20:
Fixed in both `maintenance` and `devel`, will be in next release, can be closed, right?

kevans91, 2016-05-20:
Yup! =)

closed 2016-05-20 by kevans91