OctoPrint/OctoPrint #1325: Allow for Stream URL to be from different IP/website.

DylanB00, 2016-05-04:
As stated in the title, I was wondering if you guys could make it where you can have your MJPG Stream coming from a different source rather than the Pi. The reason I ask is because I have a Windows PC I can use for higher resolutions and framerates when doing timelapses/monitoring in general.

markwal, 2016-05-04:
This already works. You can just put a url to whatever stream you want in settings.

The only requirement is that the stream is an mjpeg-stream. So, perhaps your request is that more steam formats be supported?

DylanB00, 2016-05-06:
@markwal I can't get it to work though. The image source is called (My.Rasp.Ip:Port/ActualWebAddressStream/video.mjpeg

foosel, 2016-05-06:
Not sure I understand you correctly, but make sure to also include the http:// (or https://) in front of your stream url, otherwise the browser will interpret it as a relative url.

ntoff, 2016-05-07:
@DylanB00 why are you including your raspberry pi's ip if the stream is on another computer?

foosel, 2016-06-08:
Closing this request since as stated above, this is already possible.

closed 2016-06-08 by foosel