OctoPrint/OctoPrint #1331: Octoprint should not interrupt current print when an upload from SD card happens

zobertke, 2016-05-11:
#### What were you doing?

Was printing an object, started from the control panel and not octoprint. In the mean time I've started to upload a gcode file to the sd card through the web page.

#### What did you expect to happen?
Print is going on. File is uploaded in parallel.

#### What happened instead?
The ongoing print was interrupted and could not resume it anymore.

#### Branch & Commit or Version of OctoPrint
1.2.10 (master)
[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version
Prusa I3, Marlin 1.1RC6
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome, OSX
[If applicable, always include if unsure.]

gddeen, 2016-05-11:
Excellent idea! And maybe also warn user... "This is going to take more time that you want to spend. Do you really want to use a slow serial line to do this?"

foosel, 2016-05-11:
You started the print through the controller, not through OctoPrint. So OctoPrint didn't know a print was ongoing. IF OctoPrint knows that a print is ongoing it will not allow you to upload to SD. Apparently the firmware is not as user error resilient and happily kill a running print job from SD if a new file is sent to it, so I strongly suggest to take this up with your firmware supplier. Doing such a check on the side of the firmware should be considered mandatory.

That being said, turning that into a feature request to have OctoPrint send an `M25` on start of a SD file upload and see if that reports an error (no print ongoing) or not (possibly a print ongoing, refuse to upload). Since behaviour of `M25` when not printing is not specified generally though, that will have to be an optional feature, because it's unreliable out of the box.