OctoPrint/OctoPrint #1333: [Request] Plugin for modifying location of object in STL and other similar features.

DylanB00, 2016-05-13:
Makerbot's software comes with options to lay the object flat, change how the object lays, and allows for moving the object on the plane. Something that OctoPrint lacks is a plugin with those features. Scaling options would be great too. Adding this could allow for more than one file to be printed at a time and also allow for more customization when running a print.

markwal, 2016-05-13:
There are slicer plugins and stl viewer plugins, but so far nobody has wired them together.  There's also probably lower hanging fruit in just allowing tweaking some settings for slicing rather than choosing a profile.  But what you ask for is possible, it's just a bunch of work.

Most people use OctoPrint to print gcode files that they've sliced on their computer using a slicer that allows the controls you mention.  The slicer plugins are for when you've got a routine print that doesn't need any additional tweaking so you can use a predefined profile and the stl file as is.

donovan6000, 2016-07-14:
OctoPrint's [M33 Fio](https://github.com/donovan6000/M33-Fio) plugin allows modifying a model before it's sliced. It expands OctoPrint's standard slicing routine to include the ability to modify the slicer profile and model. Even though M33 Fio is designed for use with the Micro 3D printer, you can enable its "Not using a Micro 3D printer" option to be able to use it with any printer.

M33 Fio's model editor allows translating, scaling, rotating, cloning, cutting, merging, importing files, exporting the scene, etc. M33 Fio also expands OctoPrint's 3D file format support to include OBJ, M3D, AMF, VRML, COLLADA, and 3MF files, so you can even print multiple files at the same time even if they are different formats.
[image]