OctoPrint/OctoPrint #1336: SerialException on tool 0 select (yet tool 1 works fine...)

CasperH2O, 2016-05-15:
#### What were you doing?

Using the "Select Tool" select drop down under the Control tab. 

#### What did you expect to happen?

I expect to be able to change the extruder selection from 0 to 1 and vice versa.

#### What happened instead?

Selecting tool 1 works no problem, even when 1 is already selected. However, when selecting tool 0 a SerialException occurs and the printer is disconnected:

`Send: N11 M105*23
Recv: wait
Recv: ok 11
Recv: T:19.53 /0 B:17.83 /0 B@:0 @:0 T0:19.53 /0 @0:0 T1:20.39 /0 @1:0
Recv: wait
Recv: wait
Recv: wait
Send: N12 T0*9
Recv: ok 12
Recv: FlowMultiply:100
Send: N13 M105*21
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1533
Changing monitoring state from 'Operational' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1533'
Connection closed, closing down monitor`

Selecting Tool 1 works fine:

`Recv: wait
Recv: wait
Recv: wait
Send: N5 M105*34
Recv: ok 5
Recv: T:19.65 /0 B:17.83 /0 B@:0 @:0 T0:19.65 /0 @0:0 T1:20.51 /0 @1:0
Send: N6 T1*61
Recv: ok 6
Recv: FlowMultiply:100
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Send: N7 M105*32
Recv: ok 7
Recv: T:20.39 /0 B:18.45 /0 B@:0 @:0 T0:19.65 /0 @0:0 T1:20.39 /0 @1:0
Recv: wait`

#### Branch & Commit or Version of OctoPrint

The problem started occurring yesterday after I updated from Octoprint 1.2.9 to 1.2.11. I did not experience this problem with 1.2.9. I have not yet tried downgrading.

#### Printer model & used firmware incl. version

SpiderBot v1.0 with Repetier Firmware 0.92. I have checked the settings and I have all the options for Repetier switched on.

#### Browser and Version of Browser, Operating System running Browser

Using Firefox 46.0.1 on Windows 10.

#### Link to octoprint.log

[octoprint.zip](https://github.com/foosel/OctoPrint/files/264888/octoprint.zip)

#### Link to contents of terminal tab or serial.log

See above for relevant terminal tab output. Additionally a serial log was made (had to switch it on in config).

[serial.zip](https://github.com/foosel/OctoPrint/files/264889/serial.zip)

#### Screenshot(s) showing the problem:

Additionally I have tried the same commands from Repetierhost on Windows 10. There it works fine. See screenshot. 
[image]

CasperH2O, 2016-05-15:
I can't fathom why this ticket was marked incomplete, but I'm open to hear why. Initially I posted without the javascript console part but I have since added that.

CasperH2O, 2016-05-16:
Ok I downgraded back to 1.2.9 using the following instructions from another ticket I found:

`source ~/oprint/bin/activate
cd ~/OctoPrint
git reset --hard 1.2.8
python setup.py install
sudo service octoprint restart`

The problem is no longer reproducible. I will try 1.2.10 also to narrow down the search.

foosel, 2016-05-16:
The serial exception you saw has nothing to do with the tool change and in fact that change was already acknowledged ("ok 12") when the exception occurred. The error that happened was caused on a layer way beneath what OctoPrint can control, and hints at some sort of communication problem between the system running OctoPrint and your printer. Usually that sort of problem gets triggered by a sudden disconnect or concurrent access from multiple processes to the same serial port (like the error message is hinting at). What happened was basically the serial connection claiming there was a reply but then there not actually being one when the underlying serial library tried to read the available data. 

How reliably could you reproduce that behaviour with 1.2.11?

CasperH2O, 2016-05-21:
This is slightly embarrassing, when the error occurred I was able to reproduce it 4 times (without rebooting Octoprint or my printers controller). I have since updated again to 1.2.11 on the machine that showed the problem and now on another machine (same firmware). In both cases it works fine and the issue is not reproducible.

Seems we can close this one, although I still find it a bit odd.

closed 2016-05-21 by CasperH2O