OctoPrint/OctoPrint #1338: Disk space thresholds in settings needs unit

nicolinux, 2016-05-16:
#### What were you doing?

Tried to understand which unit size the "Disk Space Thresholds" in the settings->folders area expects.

#### What did you expect to happen?

See the unit.

#### What happened instead?

Didn't see the unit :)

#### Branch & Commit or Version of OctoPrint

1.2.11 (master)

#### Screenshot(s) showing the problem:

[Imgur](http://i.imgur.com/SHXDk8H.png)

I have read the FAQ. 👍

ntoff, 2016-05-18:
Can't you just type in something like 50MB and it'll set it to 50 megabytes, or 1GB and it'll set it to 1 gigabyte.

foosel, 2016-05-20:
You can, should be documented though. Now it is ;)

foosel, 2016-05-20:
... correction. Now it is for the gcode size thresholds, not the disk space thresholds m) reopened

foosel, 2016-05-20:
Now it's also documented for the disk space threshold.

commit 82a580f5 (2016-05-20): Add help text re units for gcode size thresholds

Closes #1338
commit 436ebd7c (2016-05-20): Add help text re units for disk space thresholds

Closes #1338
closed 2016-05-20 by foosel