OctoPrint/OctoPrint #1339: Whenever I enable Turbo mode I get 503 Service Unavailable

svh1985, 2016-05-16:
When I enable Turbo mode on my Pi2 I get 503 errors.

I've manually started OctoPrint to see whats going on:
```
source ~/oprint/bin/activate
octoprint
```

> 2016-05-16 15:19:52,275 - octoprint.server - INFO - Starting OctoPrint 1.2.11 (master branch)
> 2016-05-16 15:19:52,327 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins and installed plugin packages...
> 2016-05-16 15:19:53,501 - octoprint.plugin.core - INFO - Found 21 plugin(s) providing 21 mixin implementations, 17 hook handlers
> 2016-05-16 15:19:53,509 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
> 127.0.0.1 - - [16/May/2016 15:19:53] "GET / HTTP/1.1" 200 -
> 2016-05-16 15:19:53,743 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
> 127.0.0.1 - - [16/May/2016 15:19:53] "GET /intermediary.gif?_=1463404793840 HTTP/1.1" 200 -
> 2016-05-16 15:19:54,067 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
> 127.0.0.1 - - [16/May/2016 15:19:54] "GET /online.gif?_=1463404793840 HTTP/1.1" 404 -
> 127.0.0.1 - - [16/May/2016 15:19:54] "GET /favicon.ico HTTP/1.1" 200 -
> 127.0.0.1 - - [16/May/2016 15:19:55] "GET /intermediary.gif?_=1463404795095 HTTP/1.1" 200 -
> 127.0.0.1 - - [16/May/2016 15:19:55] "GET /online.gif?_=1463404795100 HTTP/1.1" 404 -
> 2016-05-16 15:19:55,481 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 1.1
> 2016-05-16 15:19:55,493 - octoprint.plugin.core - INFO - Initialized 21 plugin implementation(s)
> 2016-05-16 15:19:55,887 - octoprint.plugin.core - INFO - 21 plugin(s) registered with the system:
> |  Active Filters (0.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters
> |  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/announcements
> |  Autoselect Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoselect
> |  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/cura
> |  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
> |  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/discovery
> |  DisplayProgress (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayprogress
> |  Navbar Temperature Plugin (0.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
> |  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/pluginmanager
> |  PortLister (0.1.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_portlister
> |  Print History Plugin (0.9) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_printhistory
> |  Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
> |  RequestSpinner (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_requestspinner
> |  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/softwareupdate
> |  STL Viewer (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stlviewer
> |  System Command Editor (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_systemcommandeditor
> |  Telegram Notifications (1.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram
> |  Title Status (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_title_status
> |  TouchUI (0.3.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
> |  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.11-py2.7.egg/octoprint/plugins/virtual_printer
> |  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
> 2016-05-16 15:19:55,894 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
> 2016-05-16 15:19:55,904 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
> 2016-05-16 15:19:55,911 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
> 2016-05-16 15:19:55,936 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
> 127.0.0.1 - - [16/May/2016 15:19:56] "GET /online.gif?_=1463404796120 HTTP/1.1" 404 -
> 127.0.0.1 - - [16/May/2016 15:19:56] "GET /intermediary.gif?_=1463404796117 HTTP/1.1" 200 -
> 2016-05-16 15:19:57,072 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "User's OctoPrint Server" for _http._tcp
> 2016-05-16 15:19:57,085 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "User's OctoPrint Server" for _octoprint._tcp
> 2016-05-16 15:19:57,089 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance "User's OctoPrint Server" for SSDP
> 2016-05-16 15:19:57,195 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
> 2016-05-16 15:19:57,242 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
> 2016-05-16 15:19:57,247 - octoprint.plugins.stats - INFO - Printer Stats
> 2016-05-16 15:19:57,341 - octoprint.plugins.portlister - INFO - Port Lister () {}
> 2016-05-16 15:19:57,349 - octoprint.plugins.stlviewer - INFO - STL Viewer loaded!
> 2016-05-16 15:19:57,610 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'https://octopi.domain.com/'}
> Segmentation fault

After I disabled Turbo Mode (back to None) everything worked again.
It seems like the Turbo Mode is messing with the stability.

markwal, 2016-05-16:
But if your pi doesn't run reliably when overclocked, what is OctoPrint supposed to do about it?

foosel, 2016-05-16:
What @markwal said. Segmentation fault means the python process crashed. Segmentation fault taking the python process with it sounds like either python itself hiccuping or some native component loaded into python. Core OctoPrint itself does not have native components (it's pure python), and if it's one of the third party libraries freaking out here there's also nothing OctoPrint can fix about that.

closed 2016-05-16 by foosel