OctoPrint/OctoPrint #1340: [Request] Toggle Switch with Feedback

LennartPl, 2016-05-16:
It would be nice to have a "toggle button" option within the custom controls. This toggle button could be  give a feedback about the state.
For Example:
A Light Button, 

Light off -> Button Grey
 -> Click Button
Light on -> Button Yellow

I have not found a nice solution to this problem.

foosel, 2016-05-16:
The problem with this is that there is usually no way to request the state of a togglable command from the firmware. So OctoPrint would need to keep track of the state of that switch internally and that would get out of synch if a command doesn't go through or in case where the controller board doesn't reset on connect (eg printrboards) it would be a 50/50 chance to guess the correct initial state.

LennartPl, 2016-05-16:
Hm, i would use it directly with the gpio of the rpi and this state is readable. For the Firmware it is a Problem,yes.

Salandora, 2016-05-17:
Even that is a problem because octoprint is not limited to e.g. a RaspberryPi but can be used on other "computers" which uses other libraries for interfacing with the gpios.
So you would need to give many different ways which would be horrible for supporting that.
The only thing I actually could think of is a script which does what you want and can give a status feedback.
Which the could be called through octoprint.
But well I have no idea if this would be a nice solution...

gddeen, 2016-05-17:
I would assume scripts would do the work, since that is what the buttons can do, invoke a script.
So the question might be how to present a tiff/gif/jpeg/html object depending upon the results from that script.

I would assume a new library could be created for general purpose IO to all machines, but that is
way over the top (IMO).  Most people would really want to just hack anything to get it to display. (I assume)
And the "No Nice way" might mean "Short of holding the Octoprint team hostage until two icons are representing his switch."

ntoff, 2016-05-22:
>I would assume a new library could be created for general purpose IO to all machines,

Couldn't you just have the button appearance changeable based on some kind of variable? Then let python worry about the GPIO stuff. That way you could call a python script to query the gpio or receive a serial input or any kind of signal and that'd toggle the state of the button.

MoonshineSG, 2016-06-02:
I have a "switch" plugin that reads the state of a GPIO (printer power & IR lights) and allows to toggle these.  And more.... pick and choose what you need ....

See https://github.com/MoonshineSG/OctoPrint-Switch