OctoPrint/OctoPrint #1341: Continue monitoring bed temperature during cooldown

040medien, 2016-05-16:
The way I understand it, Octoprint stops polling the temperature after the print is over, unless you execute additional manual commands (e.g. print head moves).

I understand many people wait until the bed has cooled down to a certain temperature (about 35-40°C) when it is safe to remove the printed object. It would be great if I could set a target cooldown temperature and Octoprint would continue monitoring the bed temperature until the bed has cooled down to it.

If this gets implemented, it would be useful if the cooldown event was exposed in the API, so plugins could e.g. send notifications based on it.

nophead, 2016-05-16:
You can wait for the bed to cool in gcode. I use this: -

```
M190 R50            ; wait for bed to cool to 50c
```

040medien, 2016-05-17:
Thank you nophead, I've just added this to my end Gcode. That said, I think for the general user, having this a dedicated feature in the frontend would make it much more accessible.

foosel, 2016-05-17:
OctoPrint always monitors the temperature as long as it's connected to the printer (polling every 5s by default), so I don't see what you are missing here? If you want to always wait for the bed to cool down, just add the command to the gcode script for finished prints (settings - gcode scripts).

040medien, 2016-05-17:
Interesting - this is not true for my Wanhao Dupicator i3 - and from nophead's comment I take that the same is true for his printer.

I don't see any temperatures until I start printing, e.g. if I just send the following in the console: M190 S60
The printer then will send temperatures until the target temperature is reached and then stop - all I get from this point is the "Recv: wait" every second. I will get new temperatures when I manually move the head.

My Gcode usually ends with M140 S0, so from that moment on I also don't get any temperatures, so the chart stops at e.g. 60C. I'll try adding a M190 S40 to the end Gcode, let's see if this works.

nophead, 2016-05-17:
Note it should be M190 R40 if you want it to cool to 40C rather than heat to 40C.

During heat up and cool down my printer constantly sends temperatures. At other times OctoPrint sends M105 to read the tempertures and updates the graph. I don't have any of the wait nonsense.

foosel, 2016-05-17:
Ah... If you have an M140 S0 at the end octoprint will assume your printer to not accept any commands at all until that temperature is reached (the normal behalf for M140 and M190) and hence not send anything in order to not overrun the buffer. Does your firmware ever return an ok for that M140?

nophead, 2016-05-17:
M140 S0 never waits, it just turns off the heater. If it waited for the temperature to get to zero it would take forever. M140 with S only waits for the temperature to increase. M140 with R waits for the temperature to decrease.

foosel, 2016-05-17:
It should still return an ok though, which should lift the "waiting for command to complete" status within OctoPrint again. 

Also if the past years working with firmware has taught me anything it's that just because something doesn't make sense (like actually waiting for 0degC to be reached) doesn't mean it won't still happen.

nophead, 2016-05-17:
S0 is a special case that turns the heater off. 
Sn sets the target to n and waits while it is lower than the target. 
Rn turns the heater off and waits while it is higher than n.
While waiting it reports the bed temperature.

In all cases it returns ok when it has finished. At least that is what the version of Marlin I use does.

foosel, 2016-05-17:
Yes, and I'd like @040medien that it does for him too, because if not that would explain the observed behaviour

040medien, 2016-05-17:
This is what M140S0 gets me:

```
: M140 S0
Recv: ok 0
Recv: TargetBed:0
Recv: wait
Recv: wait
Recv: wait
Send: M105
Recv: ok 0
Recv: T:26.51 /0 B:59.89 /0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
```

The printer accepts commands immediately afterwards, but the temperature is not updated automatically anymore.

On that note, if I run "M190 R40", it sends one M105 after that, but nothing else. Hence nophead's suggestion doesn't work for me either.

Would you like me to file a bug report instead?

nophead, 2016-05-17:
My version of Marlin behaves very differently: -

```
Send: M105
Recv: ok T:18.3 /0.0 B:18.6 /0.0 T0:18.3 /0.0 @:0.00 B@:0.00
Send: M105
Recv: ok T:18.3 /0.0 B:18.6 /0.0 T0:18.3 /0.0 @:0.00 B@:0.00
Send: M190 S40
Recv: T:18.29 E:0 B:18.8
Recv: T:18.29 E:0 B:19.8
Recv: T:18.29 E:0 B:20.6
Recv: T:18.29 E:0 B:21.5
Recv: T:18.29 E:0 B:22.8
Recv: T:18.29 E:0 B:24.0
Recv: T:18.29 E:0 B:25.1
Recv: T:18.29 E:0 B:26.3
Recv: T:18.29 E:0 B:27.5
Recv: T:18.29 E:0 B:28.4
Recv: T:18.29 E:0 B:29.7
Recv: T:18.29 E:0 B:31.0
Recv: T:18.29 E:0 B:32.1
Recv: T:18.29 E:0 B:33.2
Recv: T:18.29 E:0 B:34.6
Recv: T:18.29 E:0 B:35.6
Recv: T:18.29 E:0 B:36.7
Recv: T:18.29 E:0 B:37.9
Recv: T:18.38 E:0 B:39.1
Recv: T:18.29 E:0 B:40.0
Recv: ok
Send: M105
Recv: ok T:18.3 /0.0 B:41.0 /40.0 T0:18.3 /0.0 @:0.00 B@:0.00
Send: M105
Recv: ok T:18.3 /0.0 B:41.0 /40.0 T0:18.3 /0.0 @:0.00 B@:0.00
Send: M105
Recv: ok T:18.3 /0.0 B:40.5 /40.0 T0:18.3 /0.0 @:0.00 B@:0.00
Send: M105
Recv: ok T:18.3 /0.0 B:39.9 /40.0 T0:18.3 /0.0 @:0.00 B@:45.35
Send: M106
Recv: ok
Send: M105
Recv: ok T:18.4 /0.0 B:40.0 /40.0 T0:18.4 /0.0 @:0.00 B@:31.92
Send: M105
Recv: ok T:18.4 /0.0 B:40.0 /40.0 T0:18.4 /0.0 @:0.00 B@:36.96
Send: M190 R30
Recv: T:18.29 E:0 B:40.0
Recv: T:18.29 E:0 B:39.9
Recv: T:18.25 E:0 B:39.7
Recv: T:18.25 E:0 B:39.5
Recv: T:18.29 E:0 B:39.3
Recv: T:18.38 E:0 B:39.1
Recv: T:18.29 E:0 B:38.8
Recv: T:18.38 E:0 B:38.6
Recv: T:18.29 E:0 B:38.5
Recv: T:18.33 E:0 B:38.2
Recv: T:18.29 E:0 B:37.9
Recv: T:18.33 E:0 B:37.6
Recv: T:18.29 E:0 B:37.6
Recv: T:18.25 E:0 B:37.3
Recv: T:18.29 E:0 B:37.0
Recv: T:18.29 E:0 B:36.8
Recv: T:18.33 E:0 B:36.7
Recv: T:18.25 E:0 B:36.4
Recv: T:18.29 E:0 B:36.2
Recv: T:18.33 E:0 B:36.0
Recv: T:18.29 E:0 B:35.8
Recv: T:18.25 E:0 B:35.6
Recv: T:18.29 E:0 B:35.4
Recv: T:18.29 E:0 B:35.2
Recv: T:18.33 E:0 B:35.0
Recv: T:18.29 E:0 B:34.8
Recv: T:18.25 E:0 B:34.6
Recv: T:18.33 E:0 B:34.4
Recv: T:18.33 E:0 B:34.2
Recv: T:18.29 E:0 B:34.0
Recv: T:18.25 E:0 B:33.8
Recv: T:18.29 E:0 B:33.6
Recv: T:18.38 E:0 B:33.5
Recv: T:18.29 E:0 B:33.2
Recv: T:18.33 E:0 B:33.1
Recv: T:18.29 E:0 B:32.9
Recv: T:18.33 E:0 B:32.8
Recv: T:18.38 E:0 B:32.6
Recv: T:18.38 E:0 B:32.5
Recv: T:18.33 E:0 B:32.3
Recv: T:18.33 E:0 B:32.2
Recv: T:18.29 E:0 B:32.1
Recv: T:18.33 E:0 B:31.9
Recv: T:18.33 E:0 B:31.8
Recv: T:18.33 E:0 B:31.7
Recv: T:18.33 E:0 B:31.5
Recv: T:18.29 E:0 B:31.4
Recv: T:18.38 E:0 B:31.4
Recv: T:18.33 E:0 B:31.1
Recv: T:18.38 E:0 B:31.1
Recv: T:18.33 E:0 B:31.0
Recv: T:18.25 E:0 B:30.8
Recv: T:18.38 E:0 B:30.7
Recv: T:18.33 E:0 B:30.6
Recv: T:18.42 E:0 B:30.5
Recv: T:18.38 E:0 B:30.4
Recv: T:18.38 E:0 B:30.4
Recv: T:18.33 E:0 B:30.3
Recv: T:18.25 E:0 B:30.2
Recv: T:18.38 E:0 B:30.1
Recv: T:18.25 E:0 B:30.1
Recv: ok
Send: M105
Recv: ok T:18.4 /0.0 B:30.0 /30.0 T0:18.4 /0.0 @:0.00 B@:3.36
Send: M107
Recv: ok
Send: M105
Recv: ok T:18.4 /0.0 B:30.0 /30.0 T0:18.4 /0.0 @:0.00 B@:10.08
```

foosel, 2016-05-17:
Go into Settings > Features and check "Disable detection of external heatups" (and everything else with a "Repetier" next to it).

ntoff, 2016-05-18:
Those waits mean he's using Repetier based firmware, Rep firmware doesn't work with "R" (only S), you need to use M116 for repetier firmware to wait for temperature changes.

http://reprap.org/wiki/G-code#M116:_Wait

This is my end gcode for repetier, it waits for the temp of the hot end to reach 65 before turning off the ATX PSU cutting power (to avoid heat creep jamming the hot end)

>M104 S65		;set extruder temperature to 65 for cooldown
M140 S0 		; turn bed temp off
M106 S0		; turn off fan
G91		; Relative Positioning
G1 Z+10 F300 		; Move Z up so not to drag on part
G90 ; Absolute positioning
G1 X220 Y100 F5000 	; X out of the way Y to middle
;M84    		; disable motors(doesn't work in repetier)
;Play a tune to signify the end of the print
M300 S500 P1000
M300 S1000 P150
;;;M109 R65		; wait for extruder to cool down(doesn't work in repetier)
M116 H0		;wait for extruder to reach 65
M104 S0		; turn off hot end
M81 		; turn off the power supply

040medien, 2016-05-18:
Thanks everyone for your help. Now that I know the right settings for my firmware, it works as expected. I'll go ahead and close this FR. Maybe something to consider for the future would be a way for Octoprint to probe the printer's firmware and suggest the right settings.

closed 2016-05-18 by 040medien