OctoPrint/OctoPrint #1344: octoprint.plugin.ProgressPlugin class plugins do not get 0% progress notify

CapnBry, 2016-05-18:
#### What were you doing?

Being so happy that Gina implemented a plugin I have wanted that I almost didn't notice that it wasn't resetting when starting a new print.

OctoPrint-DisplayProgress does a lovely job of allowing me to stand in front of the printer and see how far along we are. However, once it gets to 100%%  and new print is started, the LCD indicator stays at 100%% until the new print gets to 1%(percent).

This is tracked back into OctoPrint proper, where the notification is doled out to plugins. Specifically
https://github.com/foosel/OctoPrint/blob/master/src/octoprint/printer/standard.py#L632

```
		if progress:
			progress_int = int(progress * 100)
			if self._lastProgressReport != progress_int:
				self._lastProgressReport = progress_int
				self._reportPrintProgressToPlugins(progress_int)
```

If progress is 0 then this obviously won't fire. I believe the intent of the code is to make sure there is progress/comms, which would be `if progress is not None:` the scope of this affects other plugins too, which expect to be called at 0% and 100% and all the wee percents in between. Correct?

#### What did you expect to happen?

To see 0%% when starting a new print!

#### What happened instead?

I saw the previous print's 100%% until I saw the new print hit 1%%

#### Branch & Commit or Version of OctoPrint

Version: 1.2.11 (master branch)

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

http://imgur.com/gallery/SrmH7iG

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ. **Indeed I have%.**

foosel, 2016-05-20:
You are spot on, should just be a `None` check instead.

foosel, 2016-05-20:
Fixed on `maintenance` and `devel`, will be part of the next release. Thanks for spotting this, as always a pleasure ;)

foosel, 2017-01-11:
Nope. This is NOT fixed. And never was. And I have no idea what made me think that it was either and why I only now noticed it. [This line](https://github.com/foosel/OctoPrint/blob/maintenance/src/octoprint/printer/standard.py#L171) still also captures `0` and aborts further processing. Stupid!

@CapnBry, you should have complained! :)

CapnBry, 2017-01-11:
You seemed so confident, I didn't want to break your heart! Find a bug once, shame on me. Find a bug twice, shame on you.

Actually I thought it **was** working but you can also blame my printer, which had gotten so reliable that I'm never in there any more so I replaced the progress M117 with one that displayed the enclosure temperature from a DS18B20.

Mea culpa!

foosel, 2017-01-11:
Ah, well, it should hopefully now definitely be fixed (at least I managed to receive a 0 progress in a plugin just fine after above commit). To be rolled out with 1.3.1 :)

commit 801d6038 (2016-05-20): Also report 0% print progress to plugins

Fixes #1344
commit fb5d5682 (2017-01-11): Do not eat 0% progress

Fixes #1344 (this time for good I hope)
closed 2017-01-25 by foosel