OctoPrint/OctoPrint #1345: HTTPS certificate / Lets Encrypt support

svh1985, 2016-05-18:
Hi,

Let me start with; thanks @foosel, for this awesome piece of software! I really enjoy using it.

Here's the deal, my OctoPrint (OctoPi on Pi2) is accessible through HTTPS via an open port in my router.
I've setup a dns record so I can easily access OctoPrint.

And because I want to have my login credentials sent secured I use an Certificate, the self generated certificate that came with the installation of OctoPrint.

But because it is a self signed cert, the browser will show the well know certificate errors when it is untrusted.

Wouldn't it be nice to have to have a valid certificate.. so hence; my feature request for Lets Encrypt support 👍

ps. please accept my Transifex request so help translate OctoPrint to Dutch :)

Regards,

S.

markwal, 2016-05-19:
Since the octopi image includes haproxy and there are guides for setting up letsencrypt with haproxy, I suspect OctoPrint already works with lets encrypt.

Hopefully someone will make and share a setup script for haproxy with letsencrypt support and then we could PR it into octopi to make it easier to configure, but then it would be a feature of the octopi image, not the octoprint server.

foosel, 2016-05-19:
There's also the additional problem that let's encrypt requires your server to be reachable from the Internet under the name you are trying to register the certificate for - which makes sense if you think about what such a certificate is actually supposed to do (assert that the server with ipclaiming 1.2.3.4 claiming to be example.com in fact is example.com). 

So it's not possible to create a certificate for something like octopi.local or some other internal LAN address via let's encrypt or actually - as far as I know - any certificate authority that would get accepted by your browser without a warning. Which again is actually a security feature since such a CA would be something I'd  immediately remove from the list of trusted CAs. 

Otherwise, what @markwal said - OctoPrint itself doesn't support ssl termination, that's what tools like haproxy or nginx are way better equipped for. OctoPi ships with haproxy configured to use a self signed certificate, and it could be adjusted to also include a script for generating another one in its place. 

This here is the wrong issue tracker for this though ;)

closed 2016-05-19 by foosel

svh1985, 2016-05-19:
Thanks for your replies!

The scenario I had in my mind, would be that you use **http**://octopi.local within your local network and **https**://sub.domain.com (with valid cert) when your outside your local network (internet, requires port-forwarding in router), so the problem for certificates for .local domains shouldent be an issue. That way you only get an certificate error if you browse to https://octopi.local, just like the current situation.

foosel, 2016-05-19:
You can do that, just install the let's encrypt client, generate the certificate, change the haproxy configuration to use that instead of the self-signed one. But again - OctoPrint doesn't even do SSL termination, that's what reverse proxies are used for on e.g. OctoPi, so wrong issue tracker :)

Might also make sense to actually do the SSL termination on your router instead of OctoPi. Depends entirely on your individual setup and preferences, so it's something that doesn't make much sense to "add support for", it's just too much dependent on the individual user's setup really.

eisenhowerj, 2016-09-01:
This is a top result when searching for `octoprint ssl cert` (but Gina is correct, this is the wrong place for it) but here is everything you need to get it configured:

Step 1: install certbot
https://certbot.eff.org/#debianjessie-haproxy
Step 2+: generate certs and configure haproxy
https://www.digitalocean.com/community/tutorials/how-to-secure-haproxy-with-let-s-encrypt-on-ubuntu-14-04

pcnate, 2016-09-01:
I would suggest not exposing octoprint directly to the internet. I believe I have seen this suggestion before. You should have a reverse. This would help reduce the possibility of external forces overwhelming octoprint while a print is going and causing bad things to happen to your item... You could require authentication to occur on another machine ( even if it is also a raspberry pi, a seperate one ) before ever allowing the browser to touch your octoprint

jneilliii, 2016-09-04:
Just chiming in on this one.  I've actually successfully encrypted my server and client certificate verification using the built-in haproxy of octopi without issues.  It's been a while since I set it up, but [here](https://github.com/foosel/OctoPrint/files/453730/haproxy.cfg.txt) is my /etc/haproxy/haproxy.cfg file as example (stripped of course). I used a program from sourceforge called xca to generate the files, exported server cert as type "PEM + KEY" and certificate authority as PEM and save into the same folder. This also has user auth configured to require login prior to even touching the OctoPrint interfaces.