OctoPrint/OctoPrint #1349: [REQUEST] Ability to use YouTube stream

jediknight2, 2016-05-21:
It would be nice to be able to use an embeded YouTube link/stream for the video within Octopi.  This would allow users to live stream and use that stream instead of the stream being locked by Octopi

jediknight2, 2016-05-21:
If not using the Pi for streaming to YouTube maybe just a eay to use a YouTube link as the video within the Octopi

gddeen, 2016-05-21:
Think carefully before you throw this one away...

There are far more intricacies other than this socket and that socket.

It is a portal into the web with data people want to deliver, want protected, etc...

jediknight2, 2016-05-21:
The thing about allowing an embeded YT link would be...less usage on the Pi as the video wouldnt be served from the Pi necessarily.  I want to be able to use Octoprint and video to watch..but the ability for my clients to watch their print being made

ntoff, 2016-05-22:
Wouldn't that require a separate machine just for the webcam? If you're going that far why not just use a PC to control the printer (octoprint runs on just about any OS) and broadcast that to youtube and let your viewers just watch the stream on youtube? There's no reason for a client to even look at octoprint's window unless you want to give them some measure of control (like pan/tilt/zoom).

jediknight2, 2016-05-22:
Because printing from the Pi is nicer.  I've had lockups from my PC, times where Windows Rebooted, etc...so it's not stable enough when in a 5 hour print Windows decides it needs to reboot....even though I've told it to schedule a reboot.  The BEST scenario, is for there to be a few options with the use of the camera.  If Octoprint didn't have /dev/video0 tied up...I could use the Pi and stream to Youtube if I wanted and use the webcam screen in control to show the video.  My printer happens to be beside my computer, so using the computer to do this is fine in my scenario.  I'm browsing around files and if I can find out where the web front end is being served from I might just change the coding myself.

foosel, 2016-05-22:
> If Octoprint didn't have /dev/video0 tied up...

The thing is, it doesn't. OctoPrint doesn't even know what /dev/video0 is. For OctoPrint, the webcam only exists as an url to embed as a stream into the webpage and another url to grab individual pictures from for timelapsing. At no point does OctoPrint itself actually consume the webcam stream or  access the video device. 

Which is why what you are asking is actually something I'd rather suggest to implement in the webcam server (the SD card image OctoPi uses mjpg-streamer for that). It might be possible with some creative filter chain in gstreamer and also some output plugin for mjpg-streamer. But it's nothing that belongs into OctoPrint itself (which is platform agnostic) and I'd even strongly advise against building a plugin that takes care of it since thanks to the GIL of python and the consequences that has for multi core utilisation it would have some actual performance impact possibly leading to print artifacts. 

So in a nutshell: OctoPrint is the wrong place for that since it doesn't even consume the webcam stream and actually shouldn't. You want your webcam server or a third party tool proxying your webcam stream to take care of propagating to youtube.

closed 2016-05-22 by foosel

Grogyan, 2016-05-22:
I reckon using another Raspberry PI for YT streaming would be better, and safer.

jediknight2, 2016-05-23:
Maybe...I've side stepped it by directly changing the control file and inputting the youtube stream URL.  If I have time I might look into it further.

ntoff, 2016-05-23:
>The thing about allowing an embeded YT link would be...less usage on the Pi as the video wouldnt be served from the Pi necessarily.

I don't know which pi you have (there's 3 now) but mjpeg_streamer doesn't even make a dent in my CPU usage using an "approved" webcam that natively outputs the correct format running on a raspberry pi 2b. 

>just the ability to dictate where the URL stream comes from is really all that's needed.

I've used yawcam on my windows laptop as the feed for my raspi for testing something and it works just fine. The feed doesn't have to come from the same device octoprint is running on.

brynnb, 2016-08-08:
@jediknight2 Could you provide further info on the file you edited and the line number? Thanks!