OctoPrint/OctoPrint #1350: Z Auto Calibration twice???

pcbroch, 2016-05-21:
When using Octoprint, it appears to be doing the Z axis auto calibration twice.  I have verified and I have a single G29 in my startup gcodes.  If I remove the G29, then no Z calibration happens at all. 

When I use Cura, it behaves as expected, doing a single pass.  Any ideas?

pcbroch, 2016-05-21:
Updated ticket with info requested in the template.

foosel, 2016-05-21:
Terminal tab and/or serial log still missing. Also please provide your start of print gcode script.

pcbroch, 2016-05-21:
I have a print going right now and the starting g-codes have been flushed out from the buffer.  I'll post them when my print is done and I start a new one. 

My start gcodes are:

> G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G29        ;Run the auto bed leveling
G1 Z15.0 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E10              ;extrude 10mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}

foosel, 2016-05-21:
Thanks. Would be interesting to see if it actually does send that single G29 in there twice, hence asking for the terminal output.

pcbroch, 2016-05-21:
It does send G29 twice... really odd:

> [...]
Send: M104 S210.000000
Recv: ok
[...]
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 G21*27
Recv: ok
Send: N2 G90*18
Recv: ok
Send: N3 M82*26
Recv: ok
Send: N4 M107*33
Recv: ok
Send: N5 G28 X0 Y0*23
Recv: ok
Send: N6 G28 Z0*95
Recv: ok
Send: N6 G28 Z0*95
Recv: ok
Send: N7 G29*21
Recv: Bed x: 10 y: 10 z: 0.89
Recv: Bed x: 97 y: 10 z: 0.85
Recv: Bed x: 184 y: 10 z: 0.90
Recv: Bed x: 185 y: 76 z: 0.64
Recv: Bed x: 98 y: 76 z: 0.54
Recv: Bed x: 11 y: 76 z: 0.48
Recv: Bed x: 10 y: 142 z: 0.18
Recv: Bed x: 97 y: 142 z: 0.37
Recv: Bed x: 184 y: 142 z: 0.52
Recv: Eqn coefficients: a: 0.00 b: -0.00 d: 0.80
Recv: planeNormal x: -0.00 y: 0.00 z: 1.00
Recv: ok
Send: N8 G1 Z15.0 F{travel_speed}*0
Recv: echo:endstops hit:  Z:0.52
Recv: ok
Send: N9 G92 E0*78
Recv: ok
Send: N10 G1 F200 E10*41
Recv: ok
Send: N11 G92 E0*119
Recv: ok
Send: N12 G1 F{travel_speed}*91
Recv: ok
Send: N13 M117 Printing...*41
Recv: ok
Send: N14 G90*37
Recv: ok
Send: N15 M82*45
Recv: ok
Send: N16 M106 S0*80
Recv: ok
Send: N17 M140 S63*102
Recv: ok
Send: N18 M190 S63*100
Recv: T:209.96 E:0 B:62.4
Recv: T:210.00 E:0 B:62.4
Recv: T:209.96 E:0 B:62.6
Recv: T:210.00 E:0 B:62.7
Recv: T:210.00 E:0 B:62.7
Recv: T:210.00 E:0 B:62.8
Recv: T:210.00 E:0 B:62.9
Recv: T:210.00 E:0 B:62.9
Recv: ok
Send: N19 M104 S210 T0*26
Recv: ok
Send: N20 M109 S210 T0*29
Recv: T:210.0 E:0 W:8
Recv: T:210.0 E:0 W:7
Recv: T:210.0 E:0 W:6
Recv: T:210.1 E:0 W:5
Recv: T:210.2 E:0 W:4
Recv: T:210.4 E:0 W:3
Recv: T:210.4 E:0 W:2
Recv: T:210.8 E:0 W:1
Recv: T:210.8 E:0 W:0
Recv: ok
Send: N21 G28 X0 Y0*33
Recv: ok
Send: N22 G29*34
Recv: Bed x: 10 y: 10 z: 0.47
Recv: Bed x: 97 y: 10 z: 0.43
Recv: Bed x: 184 y: 10 z: 0.48
Recv: Bed x: 185 y: 76 z: 0.22
Recv: Bed x: 98 y: 76 z: 0.11
Recv: Bed x: 11 y: 76 z: 0.05
Recv: Bed x: 10 y: 142 z: -0.24
Recv: Bed x: 97 y: 142 z: -0.06
Recv: Bed x: 184 y: 142 z: 0.10
Recv: Eqn coefficients: a: 0.00 b: -0.00 d: 0.38
Recv: planeNormal x: -0.00 y: 0.00 z: 1.00
Recv: ok
Send: N23 G92 E0*118
Recv: echo:endstops hit:  Z:0.10
Recv: ok
Send: N24 G1 E-1.0000 F1800*38
Recv: ok
Send: N25 G1 Z0.090 F1000*37
Recv: ok
Send: N26 T0*14
Recv: echo:Active Extruder: 0
Recv: ok
Send: N27 G1 X13.867 Y40.643 F3600*113
Recv: ok
Send: N28 G1 E0.0000 F540*62
Recv: ok
Send: N29 G92 E0*124
Recv: ok
Send: N30 G1 X14.601 Y40.446 E0.0134 F1500*5
Recv: ok

foosel, 2016-05-21:
Look closely though. Second one is from the file. First one is from the printing start gcode script you set in OctoPrint, second one probably from the start gcode in your slicer. Only have one *or* the other. This is not a bug, this is the combination of your OctoPrint configuration and your file to be printed containing two times the z level code.

closed 2016-05-21 by foosel

pcbroch, 2016-05-21:
Uggggghhh!   I'm an idiot!  Sorry!

nophead, 2016-05-22:
Why are there two line sixes and why did the firmware not reject the second
one?

Why are some lines italicized?

foosel, 2016-05-22:
`*` in markdown italicises and @pcbroch did a block quote instead of a code block, so the `*`s got read as markdown and caused partial italics. Double line I took as a copy paste error considering the firmware didn't complain.