OctoPrint/OctoPrint #1351: Print pause with input GPIO

temuccio, 2016-05-21:
Hello, 
it is possible to send pause command at printer when an input of GPIO change status.
With this mode I can control il filament is present or it is finished and, in this case, I don't lose the print.

temuccio, 2016-05-31:
For this request I think of resolve in this mode:
usage events to trigger the Z change.
Every Z change the system execute a control of input GPIO and if the input say that filament is end, the printer go on pause status.
In this case, if the concept is valid, how I trigger an action if true/false? In this page (http://docs.octoprint.org/en/master/events/index.html) I can't find it.

FlyByPC, 2016-05-31:
There does appear to be a "ZCHANGE" event on the page listed above. If code could be hooked in here to poll a GPIO and optionally trigger a PAUSE event if it was high (or low), that would do it. You could hook up a filament-end switch with a pull-up or pull-down resistor, and could avoid a failed print.
With an ESP8266 Thing or similar, you could even have the sensor send you a text or email.

FlyByPC, 2016-05-31:
You'd just need to ensure there's always enough filament between the sensor and extruder to complete any one Z layer, even in the worst case. (Suppose the filament triggers the sensor just after the ZCHANGE event.)

temuccio, 2016-05-31:
Yes FlyByPC, the idea is it this, but I don't understand how put in pause.
For example, if this is the config.yaml, I suppose to write:
events:
   enabled: True
   subscriptions:
   -event: ZChange
    command: "check input XX and if == 0 pause"
    type: system

How I write a condition in command line?

FlyByPC, 2016-05-31:
I don't know Python, although it's on my list to learn. It looks like you can "fire" events, which are seen by anything "subscribed" to them. [Here's the code in events.py](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/events.py).

Maybe it's as easy as adding a GPIO check and conditional pause event to the ZCHANGE event?

In Arduino C, it would read something like:

void event_ZCHANGE{

//... rest of code here

if (digitalRead(FILAMENTPIN) == LOW){fireEvent(PAUSE);}

}//event_ZCHANGE

MoonshineSG, 2016-06-02:
the best way to do it is via a custom plugin that would get a "nudge" from the GPIO and issue a "pause" and some kind of notification (play a sound?) ... 

what kind of filament sensor are we taking about ?

foosel, 2016-06-02:
You have two options here:

1. Have an external program listen on the GPIO input (e.g. by using [RPi.GPIO](http://raspi.tv/2013/how-to-use-interrupts-with-python-on-the-raspberry-pi-and-rpi-gpio-part-2)), in case of a GPIO event have it send a pause command to [OctoPrint's REST API](http://docs.octoprint.org/en/master/api/job.html#post--api-job).

2. Basically the same, but built into an [OctoPrint plugin](http://docs.octoprint.org/en/master/plugins/index.html) which can then utilize the injected [`self._printer` property](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface) (also see [Injected Properties](http://docs.octoprint.org/en/master/plugins/concepts.html#injected-properties)) to achieve the same thing.

The event hooks are the wrong approach - a GPIO input from a filament sensor won't trigger an event in OctoPrint. 

Also keep in mind that this is functionality that will NOT be added to OctoPrint proper, since OctoPrint is platform agnostic and runs on anything under the sun that runs Python 2.7 - that doesn't necessarily have to be a Pi or anything else with GPIO for that matter. So plugin-only (or external) it is.

MoonshineSG, 2016-06-02:
@temuccio: I put together a quick plugin. Change it as you need ... 
https://github.com/MoonshineSG/Octoprint-Filament

FlyByPC, 2016-06-02:
Thanks! The API -- and/or the plugin -- is probably just what I'm looking for. 

Gina -- would it be possible to add this to the next version of OctoPi, since that's targeted for RPi?

foosel, 2016-06-02:
@FlyByPC Bundling an existing universally applicable plugin might be possible, but then again it would make more sense to have people just install whatever they need through the plugin repository (that's what it's there for). Most people don't connect stuff to GPIO, so having it included by default adds unnecessary ballast.

FlyByPC, 2016-06-02:
That makes sense. I didn't know about the API -- and that alone should make a solution pretty straightforward. Thanks.

temuccio, 2016-06-05:
@MoonshineSG thanks for the plugin, the next week I cat test it. For the filament sensor , I use a sensor's paper in the common printer (in attach two foto for the idea). It's works at 3.3V ad is it compatible with raspberry.
[image]
[image]

MoonshineSG, 2016-06-29:
a clever approach using an optical sensor (using the sample code above, since then promoted to full fledge plugin):  http://www.thingiverse.com/thing:1646220

foosel, 2017-01-27:
Considering that the intent behind this request was a filament sensor and for that there's now a plugin available, I'd say this can be considered resolved.

closed 2017-01-27 by foosel

FlyByPC, 2017-01-27:
That sounds good. Thanks!

  _____  

(#1351)

<https://github.com/foosel/OctoPrint/issues/1351#issuecomment-275725764> it
on GitHub, or mute
A8J24bRks5rWiwSgaJpZM4Ij3b9> the thread.
<https://github.com/notifications/beacon/AD3fKDWLoaNsT-yyRvQ59nQYxLGsdQvZks5
rWiwSgaJpZM4Ij3b9.gif>

jypma, 2018-04-13:
I'm using the "pause print at Z layer" plugin for Cura, that allows for e.g.  filament change at a certain height. That plugin auto-pauses the print, requiring a "Resume" click in the Octoprint UI to resume the print. My printer (Anet A8 / Marlin) doesn't seem to have a menu option to resume the print, so I have to do if from the web for now.

Does that mean a GPIO pause/resume function makes sense after all? GPIO stop may as well, as a sort of "emergency stop" function.