OctoPrint/OctoPrint #1353: [Request] More control over what's accessible for non-authenticated users

b519259, 2016-05-23:
Hi.

I think there should be an option which would limit the whole UI for not logged in users to a simple login form. I just want to be able not to share any other info with anyone not allowed.

Now, I've read a couple of similar threads and in most of them there was a question from @foosel "why wouldn't you give OctoPi address to strangers". Well, the answer is "I wouldn't" and the additional answer is "I don't have to, because [Shodan](https://www.shodan.io/)" :)

That would be a bit more than the illusion of security for people who can't set up VPN ("because of reasons"), or don't want to have to connect to VPN before connection to OctoPrint from outside. Yes, I know that wouldn't secure camera stream and still nothing changes with possible DDoS attack etc., because it's not up to OctoPi.
But apart from that, I don't want anyone to see my gcode files and even printer's temperature, just because :)

Regards.

foosel, 2016-05-23:
Duplicate of #1110.

> in most of them there was a question from @foosel "why wouldn't you give OctoPi address to strangers"

Nope. I'm pretty sure I've always said "don't give your address to strangers you don't trust and ideally don't have your stuff publicly available anyhow", and if I didn't, it was probably rather due to some misunderstanding prone phrasing. In fact I've always told people to please not only depend on OctoPrint's built in access control but ideally use tunneling or at the very least further web server based access control to lock down stuff. But of course nobody wants to listen to this because effort.

closed 2016-05-23 by foosel

b519259, 2016-05-23:
Well, I see my request is just closed because I've made a mistake and wrote "wouldn't" instead of "would", which I wanted to write. Yes, my bad, I'm pretty sure the rest of the text confirms that it was a typo.
But thanks you didn't comment on the actual request, just commented almost off-topic...

foosel, 2016-05-23:
No, actually it was closed because it is a duplicate of #1110 which is an already existing request for more granular access control, including allowing users to further lock down what kind of functionality is available to anonymous users. Having multiple requests with the same or sub goals makes tracking of tasks significantly harder and hence duplicates get closed. That's nothing personal.

And sorry for the misunderstanding of your typo'd quote of mine, I'm having a lot on my hands at the moment and that this might be a typo on your part instead of a full blown misinterpretation did pass me by.

b519259, 2016-05-23:
I've seen #1110 and because it's a quite complex task, I thought I'd request something small which probably could be connected to it. 
Thanks for clarification and sorry for misunderstanding. I hope you'll find some time to get done everything you want, fingers crossed.

foosel, 2016-05-23:
There is something possible with the development version that pretty much does what you want, just not with current stable. If you feel up to playing with the `devel` branch, take a look at https://github.com/OctoPrint/OctoPrint-ForceLogin