OctoPrint/OctoPrint #1356: [Brainstorming] Addition of feature to schedule prints

Javierma, 2016-05-29:
Hi,

I have developed a feature (the last one for my final degree project) that lets the user schedule a print in order to be done when desired that can be found in branch printing_schedule at [my user's GitHub](https://github.com/javierma/OctoPrint-TFG) and is ready for pull request, but I would like first to indicate here as it includes lots of code and there is also a doubt that I would like to ask, which is if it is possible to integrate as a plugin as it requires changes in some of the existing files and I am not totally sure about how does setup.py file works.

What can be done with this feature?
It lets the user programming when a print must start, being able to set single or repetitive tasks that can be done daily, weekly and monthly as long as it does not conflicts with an already scheduled job. In order to avoid errors, it takes the most reliable time, so it will be less probable to have errors if a print has already being done as it takes into account the time spent to heat both bed and nozzle. A scheduled job will run as long as OctoPrint is running.

In which OS can it be deployed?
If I am not wrong, OctoPrint can be run in Linux, Windows and Mac OS. This feature works under Linux as it makes use of the user's crontab, which will be the one where OctoPrint runs.

What was done to make this feature work?
-If user is going to start a print, it checks if a scheduled one would not start and warns user before start printing, giving the priority to the user.

-Gets information about which file is going to be printed and when to determine if a conflict would exist. In that case, the user will be informed and print cannot be schedule, avoiding the possibility of two prints at the same time.

-Creates a specific user that will be the one that  logs in and orders the print. For security reasons, this user will only be able to log in within the same host that runs OctoPrint and web access will not be available for the user created. This user can be removed via settings but password or user settings will not have the possibility of being changed as it would not work as it should.

-Instructions were added to README as by default Raspbian has not the system clock properly configured

If there is a doubt or a comment about it, please let me know.

Thanks,

Javier

P.S.: I also did a pull request for OctoPrint - Slic3r but had no answer or comment.

ntoff, 2016-05-30:
while I can see a use for this (scheduling a print to start when you wake up for example) you need to make it platform agnostic and not rely on a specific feature unique only to Linux (or any single OS) as octoprint is designed to run on multiple hardware and software configurations.

You would also have to make absolutely sure that if a job is to be scheduled the user has removed any previous prints from the bed. Having the printer automatically print things could lead to catastrophic failure if they forget a job starts while the previous job is still attached to the build platform. Anywhere from a stringy mess to actual broken hardware.

foosel, 2016-05-30:
> you need to make it platform agnostic and not rely on a specific feature unique only to Linux (or any single OS) as octoprint is designed to run on multiple hardware and software configurations.

This. Depending on a system service for this is the wrong approach. I also don't see why a change to OctoPrint's core application would be needed to achieve such a feature. Should be perfectly achievable using something like a Python based cron-like scheduler ([apscheduler](https://apscheduler.readthedocs.io/en/latest/) comes to mind) within a plugin. For user feedback and conflict detection hooking into the available system events via a frontend component and wrapping existing internal code (e.g. print starting code) should be a good way. All without having to create a system user, having to depend on OS specific services or even having to modify OctoPrint itself.

Javierma, 2016-06-01:
@ntoff I see your point about possible failures, but I do not agree in some points.In the end, the user has the chance of choosing if the print is a single one or to be repeated,so he or she can decide using it or not with its possible risks. About making sure that the las print has been removed, maybe I am wrong but I don't see how can it be detected without 'disabling' the main point of the feature, which would be to start a print without user intervention (I suppose the printer is ready and user is conscious about what was programmed). I must say as well that I thought about repeated tasks for shops where printers are used for printing pieces to be sold (for example, to build a 3D printer) and printers (I suppose that for now not so many) that are able to remove the printed pieces. Maybe a warning message could be a solution to avoid failures.

@foosel Most of the code was developed in separate files, but for example if a user is going to start a print that wouldn't allow a scheduled job to be executed, a dialog would appear warning the user and asking to continue or not. To achieve that part, small changes need to be aplied in files.js (same file that had some changes for the feature of warning if something was going to be printed outside of the bed) and printerState.js, being the reason why I asked if it would be possible to make the feature a plugin or not (for what I see about wrapping existing internal code it is possible).

I will try to have a look at apscheduler (didn't know about that one) but for now I don't know if I will be able to apply changes to make it usable under any OS in the near future. Anyway, thanks for your help.

foosel, 2017-03-08:
I think this can be closed? As mentioned above, platform specifics sadly are a no-go, and implementation  as a plugin should in principle be possible.

closed 2017-03-08 by foosel