OctoPrint/OctoPrint #1359: Octoprint ignores baudrate setting

ozzyman, 2016-06-01:
----

#### What were you doing?

Connecting to Reprap Pro from Raspberry Pi

#### What did you expect to happen?

Printer connects

#### What happened instead?

Octoprint tries to connect at 250000 baud via /dev/ttyUSB0 (port is correct), not 115200 as specified in settings . Printer returns garbage (as would be expected, the baudrate is incorrect), octoprint disconnects

#### Branch & Commit or Version of OctoPrint

Version: 1.2.11 (master branch) 

#### Printer model & used firmware incl. version

RepRap Pro running Marlin. Unsure of version (can't get printer to connect to check)

#### Browser and Version of Browser, Operating System running Browser

Firefox 46.0.1 under Ubuntu 14.04

#### Link to octoprint.log

http://pastebin.com/2X2rdixp

#### Link to contents of terminal tab or serial.log

http://pastebin.com/hjKfeUYa

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2016-06-01:
[image]

That part there also showed 115200 when that happened?

ozzyman, 2016-06-01:
I don't know. It seems to have fixed itself after I power cycled it. Simply restarting Octoprint didn't solve the problem, and neither did rebooting the Pi, so I turned it off and went to have dinner.  When I came back and powered it up, it works as it should. I will play with it and see if I can recreate the fault. May be a hotplug issue with the USB port??

ntoff, 2016-06-03:
I find weird quirky things like this happen to me too like pip will randomly break and need a reboot or sometimes my pi will just decide to forget how to connect to the wifi network and putting the adapter in another usb port *might* fix it, or it might not. What kind of power supply are you using?

ozzyman, 2016-06-03:
I have a Chinese 2A plugpack for the Pi and a 20A 12V supply for the printer. Interesting that you mention the WiFi issues. I have had the same problem but only in the past 6 months or so. I'm running on a Pi model A and for a couple of years it performed faultlessly with an Edimax WiFi dongle (realtek RTL8188CUS) running a standard Raspbian install with Pronterface. Then for some reason it would only connect to the router every 3rd or 4th reboot, sometimes not at all. 

After much messing about with network settings I have settled on a TPLink dongle that seems to work every time (so far!). Curious thing is that the TPLink dongle is also an RTL8188 chip, although it seems to be an RTL8188EU not an RTL8188CUS. "lsusb" only shows it as "ID 0bda:8179 Realtek Semiconductor Corp", I had to scan through dmesg to find out that it is actually a 8188EU chip.

Generally, to set Octopi up for WiFi, edit /boot/octopi-network.txt and uncomment the following lines:

## WPA/WPA2 secured
iface wlan0-octopi inet manual
    wpa-ssid "<Your SSID>"
    wpa-psk "<YourNetworkPassphrase>"

It is VITAL that the SSID and Passphrase in octopi-network.txt match the ones on the router EXACTLY. I found that my router had capitalised the first letter of the SSID, so "Mynetwork" works where "mynetwork" didn't.

foosel, 2016-06-07:
@ozzyman since you can't reproduce it (and I can't either) - ok if we close this now?

foosel, 2016-06-09:
I'll just consider this a yes then ;)

closed 2016-06-09 by foosel