OctoPrint/OctoPrint #1361: Cura plugin profile update overwrite fails

vuokko, 2016-06-04:
The slicing profiles should be reloaded (preferrably from disk) when user clicks slice and dialog pops up. 
----

I found corner case which I ended up printing with some default settings. And I couldn't find any error messages. Reloading Octoprint page resolves this issue.

#### What were you doing?

I was copying known good better quality print settings to Cura from slic3r. I imported the profile to Cura plugin and printed a small object. Everything was good except I hadn't changed filament size. Quick fix in cura, export and import the profile hoping the profile would overwrite itself. From the file list I selected the object.stl and sliced it with that profile. and hit print.

#### What did you expect to happen?

Get the new settings to use.

#### What happened instead?

The print was started with bed zero, heater 220 and the print looked awful (underextrusion).

#### Branch & Commit or Version of OctoPrint
Version 1.2.11 (master branch)

commit e79703ba6eec1ecdfa21cf8f2a9efbc1eb6120e3
Author: Gina Häußge <osd@foosel.net>
Date:   Wed May 4 13:36:06 2016 +0200

    Preparing release of 1.2.11

#### Printer model & used firmware incl. version
Reprap P3Steel, Mega, Marlin RC6, RAMPS

#### Browser and Version of Browser, Operating System running Browser
Firefox 46.0.1, Linux
[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]
Sorry, I have 2MB of utf8 warnings:
2016-06-04 19:02:33,600 - octoprint.server.util.sockjs - WARNING - Could not send message to client 192.168.1.130: 'utf8' codec can't decode byte 0xb3 in position 26: invalid start byte

From Octoprint.log failed slicing:

Failed slice:
2016-06-04 15:48:45,644 - octoprint.plugins.cura - INFO - Running u'/usr/local/bin/cura_engine -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=10 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n -s retractionAmountExtruderSwitch=16500 -s supportXYDistance=700 -s insetXSpeed=50 -s retractionZHop=0 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n -s retractionSpeed=40 -s filamentFlow=100 -s infillOverlap=15 -s inset0Speed=50 -s coolHeadLift=0 -s extrusionWidth=400.0 -s upSkinCount=4 -s initialLayerSpeed=20 -s minimalLayerTime=5 -s infillSpeed=50 -s supportExtruder=-1 -s fanSpeedMax=100 -s enableCombing=0 -s fanSpeedMin=100 -s supportZDistance=150 -s supportEverywhere=0 -s filamentDiameter=2850 -s initialLayerThickness=300 -s supportAngle=-1 -s fanFullOnLayerNr=1 -s layerThickness=200 -s endCode=;End GCode\nM104 S0                     ;extruder heater off\nM140 S0                     ;heated bed heater off (if you have it)\n\nG91                                    ;relative positioning\nG1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure\nG1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more\nG28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way\n\nM84                         ;steppers off\nG90                         ;absolute positioning\n;CURA_OCTO_PROFILE_STRING:eNrtWV1v2zYUfRX6K/jWFks9SbabtIZe1iV5aYACybA2S0HQEm1xoUiBpOw4Qf57L0lJlmt7zboOXTP5wUaky/t57uElU5A0Z4LijJYmT+JwHJSKCYMNLUqqiKkUHSbh9sO4fTilWfeFfS61OV8yk+bHN0ZVGVV4nsqMJhP/FE2pWVIqkMkpSiulqDCI1qKIiMy9EPCkfXqAljkscBaZmCPQk6Oi4oaVnLZSevBkcpEzjaw1BL8ky2iGyMyAXqvz4pl4/iRoxDEpZCVMEgUzIrAuKURSkJskCsOgqPOSgm8gyTS+pUomJ4RrGuiqLKUy+BZnTBsiUoh6EI0DKrI60j+eTo4hjtM38MeVOIvCEToP0a7PpA08p8T6KWczu2IU7l3hBDPIYtZZg56xGVrJCuVkAbGb51fiSpy+itADPhNFOTEM1kHpmGFSQI5hdYSOX0ToZBiGX1htFEmNy/CMcVLYehI0ZQZ8nElFEWczVzZXV3l7y+kBMhKBWUo0RVoWFGJwr0tFtQYcOfOXP4WDMTgxRu9fxCH6YL9O7sDagnJfsHs0KSQ4fomqsrZpAdS41LpDFwCfQjq98RF6D8rCLyTF6X3/8wfracEEgupqI0t9AP46VyH5mUWZrEzj/ZKsbN7Pjkb79WpDS+gW7St9+mq/HxMy1ZJX5rO6TO5KJSEyirWBhpjfX4mnB2gbcRfh3wWdXRT1SO2R+jiRahfFPbx7ePfw3lg07Hui74l/sSc+BvqawbwMjrKCcMypmMNpIxqHgzCAQBQnJc4qwv0YnUrJrSzmZAWzt2EFTcbt0G3lsMOlBpvJVJq8Vs/dwF6P9HWSQaQZ80eDcdCtTm1/qljRXRqHQVMYGO/hFzA8SqykgoOEl9QlSSG2ZAhP7dlhVnGOc8rmuQHJsZdk9uQwIynFJmfptQDowMv40L+dArY2XgyDJQEt3UdHARMCMqBzyhunrSNbp7Ekju1zn4Y5KZ1jdcISIQX1NglT9iU4EbfpNKuSJnPFskBO/6SpwZqJa2elkbhZdY85h9vWO9mB5nARu8rpJG7PUUuWtedL+l86HNYksD4dujgKouZMJGObbGAfECkwyXLqIOcSCrgzsqgh6kvjatAeJd0x0oWGZ5wspEoULRUpgyUrofJySVV9nFzjwYLLZyrqFICIOafJS1cTyVmGve3kQlX0c6itNYSDUbcJUllMLWTdIu91C9g4kPIWUJ0zyrPaqa0esCd+GxwVZMohPG99baBp7nVrgsOx1cNxRgVQxMr31g3OpVIMlOBKuP60sAcYYuKVbplO4sFRczHhc+s616mu6QOaGf5W2tQVcTnAkBIDiYdKgDgTbsG6jZr8uscAZWqVbEaEPWFAOcaduuEFEF/hxH1BgHm38pHL0lspmSKc3dKm2msR78qojsXHPONyWTtcQ2yTJDrLfSWanO1P67S5tgBu1xi6FttbGtgYdlDLVu7t7Y8Psmbjjjcv11Q9AwU+hWHXRUfWzTXLS0hiTjT2I4K9OGqx1nEe8GPRsrAbiCUEC9paTHJITsMxbVDEcl5z63LOWWqvfMxrdJeR1b39NhR+7C4Ce9HkF6JZijQ1lir0a/TWRoV8K8Cabmfco98hhyBzt8nM9+gEEAOPu9i2us8i2C/P73bei8Es8JuAJnQbPmzawD52IoLxaCnctNSRRbaLnb7wVavvobq29ZzG0XpugNpA/AvCK6o3Nvg9+/nZUbweDmiHiq3dZkXRzLqH6zkCyuKZ2E1eRPiJYmPC2TvFeKnLsDPrXG7L+CksApLeM3e5ka2mb5TZ5ETjovCzZ4yOd8w2E3vL59bVcWbIDyvO1kkMw+bxcPeMjIZFYSct2wkIPEyvv8IMInPCfGCfhQRweAejXLvJFQBEMofJVKC3b35FOlWwadoaRIfoXS00GAzqE8lja4vRN2gL3172mLWtKv46XeF379YfpwUv9pz+mjnQ+AaJYeJrOOef9W0U7u7bKPzGjeusvbjbvQ3eu9jDB8XuBpofLfqetnraesy0Ff+Paaun7J6ye8r+WsqOvyFl9/T/neh/+CAaGIH7j48CH7b1DR9l7P3W1299f7H1ffRXofa/btj+Q3HXzWalKZYlFXjKjPYCnwDVTIal\n -s minimalExtrusionBeforeRetraction=20 -s retractionMinimalDistance=1500 -s skirtMinLength=150000 -s objectSink=0 -s retractionAmount=4500 -s skirtLineCount=1 -s startCode=M109 T0 S220.0\nT0\n;Sliced at: Sat 04-06-2016 15:48:45\n;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 20\n;M190 S?print_bed_temperature? ;Uncomment to add your own bed temperature line\n;M109 S220 ;Uncomment to add your own temperature line\nG21        ;metric values\nG90        ;absolute positioning\nM82        ;set extruder to absolute mode\nM107       ;start with the fan off\n\nG28 X0 Y0  ;move X/Y to min endstops\nG28 Z0     ;move Z to min endstops\n\nG1 Z15.0 F9000 ;move the platform down 15mm\n\nG92 E0                  ;zero the extruded length\nG1 F200 E3              ;extrude 3mm of feed stock\nG92 E0                  ;zero the extruded length again\nG1 F9000\n;Put printing message on LCD screen\nM117 Printing...\n -s skirtDistance=3000 -s printSpeed=50 -s fixHorrible=1 -s layer0extrusionWidth=400 -s moveSpeed=150 -s supportLineDistance=2666 -s posx=100000 -s posy=137500 -s sparseInfillLineDistance=2000 -s insetCount=2 -s downSkinCount=4 -s multiVolumeOverlap=150 -o /tmp/tmpeue7j7.gco /home/pi/.octoprint/uploads/control.stl' in /usr/local/bin

And slicing using good profile:
2016-06-04 19:05:06,742 - octoprint.plugins.cura - INFO - Running u'/usr/local/bin/cura_engine -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=10 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n -s retractionAmountExtruderSwitch=16500 -s supportXYDistance=700 -s insetXSpeed=60 -s retractionZHop=0 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n -s retractionSpeed=40 -s filamentFlow=103 -s infillOverlap=15 -s inset0Speed=60 -s coolHeadLift=0 -s extrusionWidth=400.0 -s upSkinCount=7 -s initialLayerSpeed=20 -s minimalLayerTime=8 -s infillSpeed=60 -s supportExtruder=-1 -s fanSpeedMax=100 -s supportType=1 -s enableCombing=1 -s fanSpeedMin=25 -s supportZDistance=150 -s supportEverywhere=0 -s filamentDiameter=1750 -s initialLayerThickness=200 -s supportAngle=60 -s fanFullOnLayerNr=1 -s layerThickness=100 -s endCode=;End GCode\nM104 S0                     ;extruder heater off\nM140 S0                     ;heated bed heater off (if you have it)\nG91                                    ;relative positioning\nG1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure\nG1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more\nG28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way\nM84                         ;steppers off\nG90                         ;absolute positioning\n;CURA_OCTO_PROFILE_STRING:eNrtWltv2zYUfjX6K/jWFks9SbabtIZe1iV5aYECybA2SyHQ0pHFRSIFk [... cut ...]  F200 E3              ;extrude 3mm of feed stock\nG92 E0                  ;zero the extruded length again\nG1 F9000\n;Put printing message on LCD screen\nM117 Printing...\n -s skirtDistance=0 -s printSpeed=60 -s fixHorrible=1 -s layer0extrusionWidth=400 -s moveSpeed=150 -s supportLineDistance=2666 -s posx=100000 -s posy=137500 -s sparseInfillLineDistance=1600 -s insetCount=3 -s downSkinCount=7 -s multiVolumeOverlap=150 -o /tmp/tmpmBpx9g.gco /home/pi/.octoprint/uploads/stop.stl' in /usr/local/bin

foosel, 2016-06-06:
I cannot reproduce this. What I did:

  * created two slicing profiles named identically, `cura_1361.ini`, one with 2.85mm filament width (`orig/cura_1361.ini`), the other with 1.75mm (`modified/cura_1361.ini`)
  * opened OctoPrint 1.2.11, Settings, Cura Engine, imported `orig/cura_1361.ini`
  * sliced an STL with `cura_1361` profile, verified generated log line contained `-s filamentDiameter=2850`
  * opened Settings, Cura Engine, imported `modified/cura_1361.ini`
  * sliced an STL with `cura_1361` profile, verified generated log line contained `-s filamentDiameter=1750`

So the updated profile was in fact used, and in fact importing a profile with the same name as an existing one will overwrite the profile data in the backend on disk. There is no caching in place here that might cause an old version of an updated profile to "linger". So whatever caused your slicing result to come out bad, it was not what you think it was.

Can you reproduce this behaviour reliably?

vuokko, 2016-06-06:
I started this again. It took me some tries to know how I made the silly mistake. One should not report bugs when they are tired but I think I found the mechanism how I got the default values.

I believe the problem is the slicing wand.

I took a profile and uploaded it as testprofile. I tried to slice with the wand already uploaded file. The profile wasn't there. Refresh populated it there and slicing worked. What I had done originally, was uploaded the fixed profile as different identifier but same name. If I upload with same identifier and name, profile is overwritten and new values are updated as it should be.

Now without refreshing I went and removed testprofile. The slicing wand had the removed testprofile available and it was good to slice. Except it had those default values. 220/ no bed.

So the slicing wand profiles aren't sync with profiles which are really available. I tested this to be true with Firefox on Linux and OS X and also with Chrome on Windows. So slicing wand should ask the profiles from server.

foosel, 2016-06-06:
The syncing issue I just fixed this morning (after noticing it while trying to reproduce your issue). The fix will be part of the next release :)

closed 2016-06-06 by foosel