OctoPrint/OctoPrint #1363: [Request] Separate webcam and time-lapse options?

jstefanop, 2016-06-07:
Currently the time-lapse video just uses a snapshot off the stream, so whatever resolution/options you have setup for the stream will be used for time-lapse. It would be great if these could be separated, and have the time-lapse code use separate options (for example i need my stream to be low-res for bandwidth purposes, but the rpi cam can do HD stills to render a nice looking 1080p time-lapse video).

foosel, 2016-06-08:
Actually, the timelapse uses a specified snapshot URL and takes whatever image it gets off of that via a simple HTTP GET for stitching together the recording. Since OctoPrint only embeds the webcam stream URL in the UI, fetches raw images from the snapshot URL and doesn't interface with the webcam directly in any way, this is actually the wrong place for this kind of request. That's something that would need to be changed in whatever webcam server is used (e.g. `mjpg-streamer`, which is the one that comes bundled with OctoPi, the SD card image that also bundles OctoPrint), so that it delivers different resolutions on stream and snapshot URL.

I can't say if that's something that would be technically feasible or not.

closed 2016-06-08 by foosel

jstefanop, 2016-06-10:
hmm ok, yea since this is specific to a rpi and its camera, i guess the best solution would be to create a plugin that uses the rpi camera API to take pictures with whatever resolutions/options you want and have a script that generates the image via your URL snapshot call.