OctoPrint/OctoPrint #1371: Request - Video Overlay of printer stats

notcranium, 2016-06-13:
When I see timelapse videos, besides the visual aspect, I like to see the details of how it was printed.   During timelapse rendering, instead of just doing an overlay of the OctoPrint logo, allow overlaying of the current stats of the printer.  

- Nozzle diameter
- Extrusion Multiplier
- Layer Height
- Shells
- Use of a Skirt, Prime Pillar, Ooze Shield or Raft
- Infill
- Nozzle Temperature
- Bed Temperature
- Fan Speed

I realize some of these stats may have to be equated from the gcode and some may not even be possible to obtain.  But it would take the timelapse video to the next level.

Mystereon, 2016-06-27:
Gstreamer 

gst-launch filesrc location="<Filepath>" ! textoverlay text="Title : Hello" valign=top halign=left ! xvimagesink

Mystereon, 2016-06-27:
http://raspi.tv/2014/overlaying-text-and-graphics-on-a-photo-and-tweeting-it-pt-5-twitter-app-series

dragonnn, 2017-03-29:
We could create for every timelapse an smi file, here is an example from my friend https://www.dropbox.com/sh/2x2s0a5fyb5xqzk/AAAh4ZrGv-aSLuXnS9C8sS8ha?dl=0
This way we don't waste CPU power to add the overlay since it is just a file, VLC or Media Player Classic reads it fine.

blindMoe, 2017-04-12:
This would be great.  I would love to be able to debug why some prints fail and also be able to tie in some extra sensors hooked up through the rPi.  Want to be able to track things like stepper motor temps and current power draw