OctoPrint/OctoPrint #1376: [Request] Larger spinning busy wheel...

itsnotlevel, 2016-06-17:
We have multiple users in our work place and would love to see something, maybe a larger busy spinning wheel...(in the center of the screen?) since you cant do anything until the last process has finished then, why not? Wheel size....maybe 300 times the size of the tiny one....

Paul

Please read the "guidelines for contributing" that are linked ^-- just
up there. Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions). Mark requests with
a [Request] prefix in the title please. Fully fill out the bug reporting
template for bug reports.

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Thank you!

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to reproduce
your issue in order to fix it and that is not possible if they don't know
what you did to get it to happen in the first place. If you encountered
a problem with specific files of any sorts, make sure to also include a link to a file
with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2016-06-17:
Sorry but: what wheel are you even talking about?

itsnotlevel, 2016-06-17:
Oh sorry, its a plugin.... 
https://github.com/OctoPrint/OctoPrint-RequestSpinner

Paul

foosel, 2016-06-17:
In that case, wrong bug tracker.

And you can in fact do things while that thing is spinning, it's merely showing background activity/server communication. Having it take up the whole screen would make the application pretty much unusable most of the time.

closed 2016-06-17 by foosel

itsnotlevel, 2016-06-17:
Good point.... Thank you

Paul